RavenSystem/esp-homekit-devices #468: Wait for the SSID forever

i3laze, 2019-08-08:
> Maybe there’ll be such a choice someday?

Sure.

_Originally posted by @RavenSystem in https://github.com/RavenSystem/esp-homekit-devices/issues/328#issuecomment-481803451_

I took down my outdoor AP for a maintenance bigger than 300 sec..
And now bunch of RC devices stated unresponsive and broadcast own SSID,

What will happen if I just power cycle them? Will they reconnect to WiFi?
How about at last implementing “wait/scan forever” as a default option?

How currently works client roaming? What if RC connected to -80 dBi AP and a -50 dBi AP comes back online?

peros550, 2019-08-18:
Latest version of [esp-wifi-config](https://github.com/maximkulkin/esp-wifi-config/issues/13) does both. It fires up an AP pretty quickly (maybe in 30sec) but at the same time it waits for the known SSID forever. I believe it combines the best from both worlds.

RavenSystem, 2019-08-25:
Security issue fixed. Now device will wait for the SSID forever.

closed 2019-08-25 by RavenSystem