RavenSystem/esp-homekit-devices #472: homekit-divices have no hostname in router

douzs, 2019-08-22:
hey,
I have a little question:
I have a lot of devices which  flashed the firmware and I can't distinguish them in the router. They have no hostname in router.
I want to konw how to give them name like RavenCore-%02X%02X%02X 
I mean the source code can change to solve the little problem ,or not?

thank you

closed 2019-11-29 by RavenSystem

peros550, 2019-11-29:
Was this included in latest version ?

RavenSystem, 2019-11-30:
No, but code is ready. I want to test right. It will be available with next version.

mafyata, 2019-12-04:
How can I change the hostname? I updated my devices and entered in setup mode to see is there a way to change it , but nothing.

RavenSystem, 2019-12-05:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect#n-custom-hostname

douzs, 2019-12-05:
@RavenSystem That's so greate! thank you

peros550, 2019-12-05:
Man! We can’t keep up with your pace! 
And I still haven’t had the chance to try the RestAPI feature :):):)):)

Thank you Jose!

macjl, 2019-12-05:
Thank you Jose for this feature. I have tested it, and it works great on the network. I can access to my devices with custom hostname.

I have noticed that on Homekit side, the devices still have the HAA-xxxxxx name in Home App. Is it wanted ?

RavenSystem, 2019-12-05:
> I have noticed that on Homekit side, the devices still have the HAA-xxxxxx name in Home App. Is it wanted ?

Yes, it is wanted. It must be unique to avoid issues with HAP, and it must be short, to use less memory. That name is used with mDNS/Bonjour service. But then you can rename HomeKit services into Home App (Internally, HAA-XXXXXX is used to manage accessory).

Other feature is that you can change device hostname in any moment, and you will keep all HomeKit settings, because mDNS name is the same always. In other case, you will must reset HomeKit ID and repair device when you change device hostname.