RavenSystem/esp-homekit-devices #473: Sonoff T1 R2 US Touch Board V1.0 flashing issue

1nutty1, 2019-08-26:
Hi all, hoping someone can help.
I am having issues getting a Sonoff T1 R2 US TOUCH BOARD V1.0 into flash mode.
Any definitive way to get this into flash mode?
[image]

pandarek, 2019-08-26:
Try this
https://user-images.githubusercontent.com/36041520/50041600-25bad980-008a-11e9-840b-86e431c12968.jpg

https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK,-EU,-and-US

1nutty1, 2019-08-26:
> Try this
> https://user-images.githubusercontent.com/36041520/50041600-25bad980-008a-11e9-840b-86e431c12968.jpg
> 
> https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK,-EU,-and-US

Thanks, will give it another go tonight when I get home.

1nutty1, 2019-08-26:
> Try this
> https://user-images.githubusercontent.com/36041520/50041600-25bad980-008a-11e9-840b-86e431c12968.jpg
> 
> https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK,-EU,-and-US

No luck. I try shorting R19 to ground and the button light stays stable but I am unable to erase the flash, I try double clicking S1 same result, I try holding the capacitive button and then pressing S1 nothing...
When I power on without shorting R19 to ground the button light is stable but if I dont it pulsates...

Any other ideas?

1nutty1, 2019-08-26:
> Try this
> https://user-images.githubusercontent.com/36041520/50041600-25bad980-008a-11e9-840b-86e431c12968.jpg
> 
> https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK,-EU,-and-US

Just had a good look at the chip, the one I have has an ESP8285... would this make any difference?

pandarek, 2019-08-27:
ESP8285 is just like ESP8266 but with 1MB built-in Flash.

You need to connect GND and GPIO0 together and then connect the power supply (from USB flasher). Then you can disconnect GPIO0.

Before uploading new software, delete the current software.
(esptool.py -p / dev / <your_ESPPort> erase_flash)

Then, set your device in flash-mode again (disconnect power, reconnect GND + GIPO0, reconnect power).

One more thing, I had a problem with USB power, too low power. (Sonoff Touch T1 EU)
I connected 3.3v and GND from external power supply (about 1A). Only TX and RX to usb flasher.

1nutty1, 2019-08-27:
> ESP8285 is just like ESP8266 but with 1MB built-in Flash.
>
> You need to connect GND and GPIO0 together and then connect the power supply (from USB flasher). Then you can disconnect GPIO0.
>
> Before [...]

Thanks a million, I will organise external power and give it another go.

1nutty1, 2019-08-28:
> can you post a photo of the other side of this board?

I will post this afternoon for you when I get home.

1nutty1, 2019-08-28:
> can you post a photo of the other side of this board?

[image]

xtremedays, 2019-08-29:
I’ve done a few of the us t1 3ch by jumping as per pic in the link. Hold it on, power up, wait 5ish seconds and disconnect the r19. Worked every time for me

walvespit, 2019-08-29:
i also did some t1 and t2 using the R19 , try both sides of it!

1nutty1, 2019-08-30:
> i also did some t1 and t2 using the R19 , try both sides of it!

Thanks for this, I have tried that to no avail.

1nutty1, 2019-08-30:
> i also did some t1 and t2 using the R19 , try both sides of it!

What TTL adapter did you use?

1nutty1, 2019-08-30:
Still pulling my hair out with this one, what adapters are you guys using?
Mine is the CH34
[image]
0

AramVartanyan, 2019-08-30:
Your wiring isn't seems to be right.
As far as I see at your picture - the programmer’s TxD (white cable) is not connected to the board. Instead it is connected to GND.
Also you have to connect the programmer’s GND to the power supply’s OUT-.

As I remember for this product:
TxD (programmer) -> RxD (board)
RxD (programmer) -> TxD (board),
but it might be the opposite. Try both.

walvespit, 2019-08-30:
> Still pulling my hair out with this one, what adapters are you guys using?
> Mine is the CH34
> [image]
> 0
you even use an external power source? wow i never needed that!

i use an CH340G adapter

1nutty1, 2019-08-30:
> > Still pulling my hair out with this one, what adapters are you guys using?
> > Mine is the CH34
> > [image]
> > 0
> > you even use an external power source? wow i never needed that!
> 
> i use an CH340G adapter

Using the same, all the other switches I have flashed have not given me this headache.
I have 7 to do for my house so will try another and see if it is maybe just this board...
Will let you know.

walvespit, 2019-08-30:
> Using the same

mine is physically different than yours but its the same chip.

im not sure if it is a good idea to use an external power source because it would use diferente GND.. 

anyway, have you ever flashed any other sonoff device yet?

1nutty1, 2019-08-30:
> > Using the same
> 
> mine is physically different than yours but its the same chip.
> 
> im not sure if it is a good idea to use an external power source because it would use diferente GND..
> 
> anyway, have you ever flashed any other sonoff device yet?

Flashed a bunch of Basics,RF s Th16 s, T1 2Ch... just tested the TTL adapter and it wont even flash a basic anymore so would appear that I need to get a new adapter?!? Rather strange that it just died tho...
Will let you know if I come right after acquiring a new adapter...

1nutty1, 2019-09-06:
As it turns out my TTL went belly up.
New TTL and all is well in the world.

Thanks for all the suggestions and help, it is greatly appreciated.

closed 2019-09-06 by 1nutty1