RavenSystem/esp-homekit-devices #475: ON/OFF switch question

kiralikbeyin, 2019-08-26:
My buttons are normal ON/OFF switches not momentry/push buttons.
[Maintained switches -- like the light switches on your wall -- stay in one state until actuated into a new one, and then remain in that state until acted upon once again. These switches might also be called toggle or ON/OFF switches.]

I am trying adv_button codes. Which configuration (X) is fine for sonoff dual, shelly1  and shelly2.5?

````
button_gpio_1 = 0; //sonoff dual
button_gpio_2 = 9; //sonoff dual

adv_button_create(button_gpio_1, true, false);
adv_button_create(button_gpio_2, true, false);

adv_button_register_callback_fn(button_gpio_1, button_simple1_intr_callback, X, NULL);
adv_button_register_callback_fn(button_gpio_2, button_simple2_intr_callback, X, NULL);
````

kiralikbeyin, 2019-08-26:
This works with dual
````
button_gpio_1 = 0;
button_gpio_2 = 9;

adv_button_create(button_gpio_1, true, false);
adv_button_create(button_gpio_2, true, false);

adv_button_register_callback_fn(button_gpio_1, button_simple1_intr_callback, 0, NULL);
adv_button_register_callback_fn(button_gpio_1, button_simple1_intr_callback, 3, NULL);
adv_button_register_callback_fn(button_gpio_2, button_simple2_intr_callback, 0, NULL);
adv_button_register_callback_fn(button_gpio_2, button_simple2_intr_callback, 3, NULL);
````

@RavenSystem There timers in adv_button lib i hope they dont make random resets becuase of very long switch state (maybe 24 hours) ?

RavenSystem, 2019-08-26:
Use 0 and 1, and not 3, for a better response:
```
adv_button_register_callback_fn(button_gpio_1, button_simple1_intr_callback, 0, NULL);
adv_button_register_callback_fn(button_gpio_1, button_simple1_intr_callback, 1, NULL);
```

kiralikbeyin, 2019-08-27:
@RavenSystem Thanks. 

For your info: 
I was using a few ```vTaskDelay(1000 / portTICK_PERIOD_MS);```  after button_simple1_intr_callback and sometimes i saw exceptions about timers.
I removed vTaskDelay's and everything is fine.

https://www.freertos.org/RTOS-software-timer.html

closed 2019-08-27 by kiralikbeyin