RavenSystem/esp-homekit-devices #476: Magic Home LED

St0nehead, 2019-08-26:
Successfully flashed Magic Home LED controller.
Model ZJ-WFMN-A v1.1.
But can’t get it to work... Current setup is:
Device type : 14
Board Type : 1

Pin G : 12
Pin W : 0
Pin R : 5
Pin B : 13

But, when i try to switch colors in Home.app, all colors getting messed up.. Green instead of red, red instead of green, purple instead of white and so on.

Please help! 🙏

RavenSystem, 2019-08-26:
It seems that your RGB pins are swapped. Set green, red and blue in your iOS device, and see which colors lights in your LED strip.

St0nehead, 2019-08-27:
I tried to swap pins, but still can’t find right order... Colors always shifted.
Which other pins can be used? 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

epoppis, 2019-08-28:
Please give a try the following config:

Pin B: 13
Pin G: 5
Pin R: 12
Pin W: 15

closed 2019-09-05 by RavenSystem