RavenSystem/esp-homekit-devices #479: Built from source

Alexivia, 2019-08-28:
Hello,

When I build the HAA and RavenCore from source, I only get a single binary file at the `/firmware` folder, and if I flash this file using the `esptool.py` the device does not seem to respond nor diffuse any hotspot... am I doing this wrong? Your build wiki is lacking some deeper explanation of what is happening, which would be useful for developers to understand the inner workings of the system.

Thank you in advance, and congratulations on the project so far!
Regards,
Alexandre

walvespit, 2019-08-29:
After Building main.bin you need to flash it together with [rboot.bin](https://github.com/SuperHouse/esp-open-rtos/raw/master/bootloader/firmware_prebuilt/rboot.bin) and [blank_config.bin](https://github.com/SuperHouse/esp-open-rtos/raw/master/bootloader/firmware_prebuilt/blank_config.bin)!

Alexivia, 2019-08-30:
Ok, that makes sense, I haven't tried that yet... These files are the ones I can find at the top of the [Installation Wiki](https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation)?

Thank you for your answer!

RavenSystem, 2019-08-30:
`make test`

Steps 4 and 5 from https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#build-instructions

Alexivia, 2019-08-31:
> After Building main.bin you need to flash it together with [rboot.bin](https://github.com/SuperHouse/esp-open-rtos/raw/master/bootloader/firmware_prebuilt/rboot.bin) and [blank_config.bin](https://github.com/SuperHouse/esp-open-rtos/raw/master/bootloader/firmware_prebuilt/blank_config.bin)
> 
> esptool.py -p /dev/<your_ESPPort> --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 
> 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 main.bin
> 
> Also remember to erase flash before !

After trying again to upload following this instructions, I made it work and the switch is now recognisable on the Home App of my iPhone! Thank you for the tips!

> `make test`
> 
> Steps 4 and 5 from https://github.com/RavenSystem/esp-homekit-devices/wiki/Build-instructions#build-instructions

I'd read this on the build manual, however I am using a Docker image with the ESP SDK, and since I'm on a Macbook, `/dev/*` devices can't be correctly passed to a Linux Docker container, thus I was unable to upload using the SDK...

I have just another question. Are there any new versions of the `rboot.bin` and `blank_config.bin` files, or these do not need update?

Thank you again. Regards.

closed 2019-08-31 by Alexivia