RavenSystem/esp-homekit-devices #482: "don't group accessories" function on iOS 13

mientki, 2019-08-29:
Is it possible to add the "don't group accessories" function to the device type 2 or 11 (switch 2ch or 3ch)? I have such a situation that the sonoff touch 2 channel switch is used for the bathroom - the first channel for the light, the second channel for the fan. By default, these accessories will be grouped in iOS 13, and turning on the second accessory will not light the tile in the home app. The only sensible solution is not to group two different types of accessories from the same device.

Bolshoylu, 2019-09-04:
+1 I join this problem too.

RavenSystem, 2019-09-05:
This is an Apple issue, I can not enable or disable characteristics grouping of same accessory.

closed 2019-09-05 by RavenSystem

mientki, 2019-09-05:
but you can create them as two separate accessories.  soon everyone was crying with mistakes.  premiere ios13 soon.

RavenSystem, 2019-09-05:
Move to HAA.

mientki, 2019-09-06:
Can you give me a template example for sonoff touch 2 gang? Will the accessories be separate?

egorist, 2019-10-27:
> Can you give me a template example for sonoff touch 2 gang? Will the accessories be separate?

{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9}]}]}