RavenSystem/esp-homekit-devices #483: I can't configure sonoff in app home after resetting wifi

ConradoP, 2019-08-30:
good night, after the setup that worked perfectly, I reset the wifi network by clicking 10 times at boot time and it worked.

But after setting up the new wifi network, which also worked perfectly, I could not add the accessory in the app home, it no longer appears.

can you help me ?

walvespit, 2019-08-30:
i have the same problem with some sonoff here ... it is conected to my network but i cannot also add to homekit, in my case i had others sonoff so i just changed it but ye it is a strange bug, i even tried to re-flash it and that did not work either, homekit found it but it says "its already added" but its not! 

I google it and someone said to reset all home app (i would have to reconfigure all my 20+ homekit devices again) and thats not great

ConradoP, 2019-09-03:
Can anyone help me?

After removing the accessory through the eve app, I can no longer add a new wifi.
The network is not available and from what I see in the log it is still connected to the current network.

how do i reset the wifi network?

ConradoP, 2019-09-03:
look that even removing in the EVE app and after restarting the sonoff the log says still connects to the same network

root@xxx:~# screen /dev/ttyUSB0 115200
ip:192.168.0.82,mask:255.255.255.0,gw:192.168.0.1
>>> wifi_config: wifi_config_sta_connect_timeout_callback: Successfully connected
RC > Creating HK accessory
RC > Creating Setup
RC > Starting HomeKit Server
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 17:C5:0A:CD:62:42
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=RavenCore-67C46F md=RavenCorepv=1.0id=17:C5:0A:CD:62:42c#=4160s#=1ff=0sf=1ci=8
                      sh=9PsfuA== Port=5556 TTL=4500

How do I reset the network?

ConradoP, 2019-09-04:
more logs

root@xxx:/sonoff# screen /dev/ttyUSB0 115200
Timer Start Failed
scandone
add 0
aid 2
cnt

connected with xxx, channel 11
dhcp client start...
ip:192.168.0.82,mask:255.255.255.0,gw:192.168.0.1
RC > Saving last states
>>> wifi_config: wifi_config_sta_connect_timeout_callback: Successfully connected
RC > Creating HK accessory
RC > Creating Setup
RC > Starting HomeKit Server
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 08:EB:75:37:C8:05
>>> HomeKit: Found admin pairing with F61027D9-E74B-4149-9BF2-C5F560316F22, disabling pair setup
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=RavenCore-67C46F md=RavenCorepv=1.0id=08:EB:75:37:C8:05c#=4160s#=1ff=0sf=0ci=8
                      sh=m2ZIyw== Port=5556 TTL=4500

ConradoP, 2019-09-11:
anyone can help me?

RavenSystem, 2019-09-28:
Logs say that your device is already paired.

closed 2019-11-07 by RavenSystem