RavenSystem/esp-homekit-devices #488: Can not pair

lexxnsk, 2019-08-30:
After some plaing with reflashing adding and removing device in HomeKit I can't even add it to the homekit right after reflashing,
11:54:52.834 -> RC > Saving last states
11:55:04.104 -> ip:10.48.20.195,mask:255.255.255.0,gw:10.48.20.1
11:55:04.375 -> >>> wifi_config: wifi_config_sta_connect_timeout_callback: Successfully connected
11:55:04.375 -> RC > Creating HK accessory
11:55:04.375 -> RC > Creating Setup
11:55:04.375 -> RC > Starting HomeKit Server
11:55:04.375 -> >>> HomeKit: Starting server
11:55:04.375 -> >>> HomeKit: Using existing accessory ID: 12:71:C1:B8:EE:E2
11:55:04.375 -> >>> HomeKit: Configuring mDNS
11:55:06.997 -> mDNS announcement: Name=RavenCore-0B989B md=RavenCorepv=1.0id=12:71:C1:B8:EE:E2c#=4160s#=1ff=0sf=1ci=8sh=mNv6TQ== Port=5556 TTL=4500
11:55:21.725 -> >>> HomeKit: Got new client connection: 4
11:55:21.725 -> >>> HomeKit: [Client 4] Pair Setup Step 1/3
11:55:35.534 -> >>> HomeKit: [Client 4] Closing client connection
11:55:36.412 -> >>> HomeKit: Got new client connection: 4
11:55:36.412 -> >>> HomeKit: [Client 4] Pair Setup Step 1/3
11:55:53.357 -> >>> HomeKit: [Client 4] Pair Setup Step 2/3
11:56:07.622 -> >>> HomeKit: [Client 4] Pair Setup Step 3/3
11:56:07.790 -> >>> HomeKit: Added pairing with 52BE63EE-6F0E-49BE-897C-9E3206C8EC80
11:56:07.858 -> >>> HomeKit: Configuring mDNS
11:56:08.432 -> mDNS announcement: Name=RavenCore-0B989B md=RavenCorepv=1.0id=12:71:C1:B8:EE:E2c#=4160s#=1ff=0sf=0ci=8sh=mNv6TQ== Port=5556 TTL=4500
11:56:08.465 -> >>> HomeKit: [Client 4] Successfully paired
11:56:08.465 -> >>> HomeKit: [Client 4] Closing client connection
11:56:28.505 -> >>> HomeKit: Got new client connection: 4
11:56:28.505 -> >>> HomeKit: [Client 4] Pair Setup Step 1/3
11:56:28.505 -> >>> HomeKit: [Client 4] Refusing to pair: already paired
11:56:28.505 -> >>> HomeKit: [Client 4] Closing client connection

Buf homekit show failed to add accessory. and thats all.

walvespit, 2019-08-30:
I have the same problem with 2 sonoffs basic here, it says its already paired, but its not!

I googled it and they said that i would have to delete my "home" config and reconfigure all devices again, thats not a great solution at all...

lexxnsk, 2019-08-30:
I’ve restarted phone and it resolved the problem. Try it

closed 2019-09-05 by RavenSystem