RavenSystem/esp-homekit-devices #491: Sonoff 4ch pro Interlock mode

simoneorsi, 2019-08-31:
Hallo to everybody. Anyone have a problem to use sonoff 4ch pro to command open close window with interlock mode.
I try to set sonoff switch in all mode but the interlock mode didn`t respond good.

First rele is on, when I switch on the second one the first one must switch off but everytime nothing happens. on to the home kit appear the first one and second one on but only the first one remain on.
If I try to test more and more time ecerytime the interlock mode function it works.

Anyone have the same problem. I will rollback to the old version (0.0.4.1)

Tghanks

simoneorsi, 2019-09-02:
In althernative it`s possible to increase thew inching time for all 4 channel for example 30sec.
In this case I can setting the sonoff in this mode and can help me a lot.
Thnaks

RavenSystem, 2019-10-07:
With HAA, you can increase inching time to 30 seconds, and much more.

closed 2019-10-13 by RavenSystem