RavenSystem/esp-homekit-devices #492: HAA vs RC

Alexivia, 2019-09-01:
From what I understand by reading your documentation, the HAA firmware is a more advanced firmware (and thus, I suppose customisable) than the RC, which seems to be a pre-configured HAA for specific Sonos, etc. devices available on the market.

That said, I would like to know if there is a way to configure an RGB/W LED strip in HAA, and if so what are the keys and values I should use in the JSON file.

Thank you. Regards.

walvespit, 2019-09-01:
HAA will be more advanced than RC but still on early releases, for now it doesnt support RGB yet

Alexivia, 2019-09-01:
> HAA will be more advanced than RC but still on early releases, for now it doesnt support RGB yet

Nice to know, thank you for the quick reply.

closed 2019-09-01 by Alexivia