RavenSystem/esp-homekit-devices #495: HAA Templates page not up to date with HAA v0.0.6

macjl, 2019-09-07:
Hello,

The Wiki page of HAA templates seems not up to date with the new release of HAA v0.0.6 : 
https://github.com/RavenSystem/esp-homekit-devices/releases
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates

I'm trying to install a new Sonoff Basic with this JSON inspired from Templates, and it doesn't work : 
{"c":{"l":13,"i":1},"a":[{"t":1,"r":[{"g":12}],"b":[{"g":0,"p":1,"t":1},{"g":2,"p":1,"t":1}]}]}

Regards,

RavenSystem, 2019-09-07:
https://twitter.com/RavenSystem/status/1169017948591329281

macjl, 2019-09-07:
Thanks for your quick reply. Is there a way to downgrade to the v0.0.5 release to be able to use old configs?

macjl, 2019-09-08:
Hello,

I was finally able to produce a valid JSon for 0.0.6 by analysing the code. Here it is :
`{"c":{"l":13,"i":1},"h":[{"t":1,"a":{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]}},"b":[{"g":0},{"g":2}]}]}`
=> Sonoff Basic with button on GPIO 2 and GPIO 0, and Relay on GPIO 12.

I could suggest to change this sentence in documentation : 

> "a" Actions:
> Actions performed on each state of the accessory. Are defined by a integer number.

I search long time to understand what it means. And in the source code, it's commented as an array which confuses me a lot. But it's just a list of states. Why not : 

> "a" Actions:
> List of actions performed on relays, at each state of accessory. Are defined by a integer number starting at 0.

macjl, 2019-09-08:
And if you want the the Sonoff S20 example:

> Sonoff S20/S26 without UART output, with status LED and working button:
`{"c":{"l":13,"i":1},"h":[{"t":2,"a":{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]}},"b":[{"g":0}]}]}`

mientki, 2019-09-11:
I need a template for sonoff touch t1 2 channel. Can You help me?

mientki, 2019-09-14:
"h":
"v":
What is this parametr?

macjl, 2019-09-17:
From Wiki :

> "h" HomeKit Accessories
> This section is mandatory. It contains an array of HomeKit accessories.

> "v": Value:
>  0: Low, normally off.
>  1: High, normally on.

mientki, 2019-09-18:
There was a small commotion. Is now the description of HAA Templates https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates and the description https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect is compatible with HAA version 0.0.7?

RavenSystem, 2019-09-18:
> There was a small commotion. Is now the description of HAA Templates https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates and the description https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect is compatible with HAA version 0.0.7?

Yes.

RavenSystem, 2019-09-18:
All documentation is updated.

closed 2019-09-18 by RavenSystem