RavenSystem/esp-homekit-devices #497: New dimmer by Shelly. Possible to support?

LyoshaDovolen, 2019-09-09:
https://shelly.cloud/wifi-smart-home-automation-shelly-dimmer/

epl2005, 2019-11-03:
Any news on this request ?

RavenSystem, 2019-11-03:
First, we need info about how it works.

RavenSystem, 2019-11-07:
If Shelly Dimmer uses PWM like Shelly RGBW2, now HAA v0.6.0 has PWM Lights support.

epl2005, 2019-11-12:
Good news ... shelly  dimmer I ordered finally it's arriving. Have You got already any confirmation  about PWM support ?

mafyata, 2019-11-12:
Read wiki. Haa support it.

epl2005, 2019-11-12:
> Read wiki. Haa support it.

If you are referring to PWM itself I know it. I was asking for Shelly dimmer fully support in reply to RavenSystem post

RavenSystem, 2019-11-14:
@epl2005 I don't know yet if Shelly Dimmer works with PWM like Shelly RGBW2.

epl2005, 2019-11-23:
> @epl2005 I don't know yet if Shelly Dimmer works with PWM like Shelly RGBW2.

I finally got mine. Is there anything I can do to verify if it’s support ? Do you have any Json config to try ? Thank you

kompak23, 2019-11-24:
Oficially, Shelly Dimmer does not use PWM modulation. It contains MOSFET transistors which conduct part of waveform. There is STM32 to control MOSFETS.  It isn't good news, we cannot use Raven software to control it under Homekit :(

epl2005, 2019-11-25:
> Oficially, Shelly Dimmer does not use PWM modulation. It contains MOSFET transistors which conduct part of waveform. There is STM32 to control MOSFETS. It isn't good news, we cannot use Raven software to control it under Homekit :(

Yes not a good news with the current HAA version ... any chance in the future ?

RavenSystem, 2019-11-26:
We need to know how esp8266 manages stm32 chip.

RavenSystem, 2019-12-10:
Bad news: https://github.com/arendst/Tasmota/issues/6914#issuecomment-562109267

pilot1981, 2019-12-10:
@RavenSystem you use TASMOTA code for your firmware?

RavenSystem, 2019-12-10:
No. Tasmota is Arduino based.

GPL71, 2019-12-11:
@RavenSystem the best would be join efforts ... draw a pcb and code ...                                        
                                                                                                                                                                  
 If people are interested work together to develop a custom pcb, it would be possible realize 220v dimmer

aleroot, 2020-01-29:
It looks like in Tasmota they figured out the way to make it work,  reverse engeneering it ... https://github.com/arendst/Tasmota/issues/6914#issuecomment-577355864

edge90, 2020-02-02:
@aleroot yeah, the two mcus communicate via uart. :)

iCarl, 2020-03-13:
Any news on this?

RavenSystem, 2020-04-25:
This must be opened.

RavenSystem, 2020-04-25:
@edge90 Shelly Dimmer uses UART from GPIO 1 and 3 of esp8266, right?

edge90, 2020-04-28:
@RavenSystem Yes, check the tasmota thread and you'll find a lot of dumps etc.

RavenSystem, 2020-04-28:
@edge90 
I am using some codes from here: https://github.com/arendst/Tasmota/issues/6914#issuecomment-575899122

And I think this is right: https://github.com/arendst/Tasmota/issues/6914#issuecomment-575747738

What do you think?

RavenSystem, 2020-04-28:
@edge90 How is checksum calculated?

closed 2020-05-19 by RavenSystem

aleroot, 2020-05-31:
If you have a look at this tasmota version, they seem to have completed the integration :

https://github.com/arendst/Tasmota/issues/6914#issuecomment-636297822

Shelly dimmer is working now with tasmota ...

aleroot, 2020-06-11:
Is there any plan to support this dimmer from Shelly ?
Has this ticket been closed because there is no plan to support it?

tokamac, 2020-07-18:
I concur, why this has been closed (for the 2nd time) without being resolved, just at the moment Tasmota guys seem to have found a working solution? IMHO a dimming module compatible with HomeKit would be a must-have, if not the best possible feature for HAA.

I see that Shelly Dimmer (first version) is listed in [Devices Database](https://github.com/RavenSystem/esp-homekit-devices/wiki/Devices-Database). But does the Shelly Dimmer **2** work with HAA? 

Shelly Dimmer 2 does not need neutral wire (first version needs it) which is a killer feature, since many switch boxes do not have neutral wires. There is no other small dimmer module working with no neutral wire on the market that I am aware of, besides Fibaro Dimmer 2 (which is Z-Wave and needs homebridge if one wants HomeKit compatibility).