RavenSystem/esp-homekit-devices #502: Help needed with Magic Home Single Channel

Olepir, 2019-09-15:
How to flash Magic Home Single Channel with RavenCore or HAA?
Can I flash it with RavenCore? If not – how to use HAA and is there template available for this LED controller?
Thanks

mafyata, 2019-09-15:
Ravencore. I have 2 magic home rgb led controllers flashed with ravencore. After flash with ravencore you can’t controll it with normal ir controller, only with phone/tablet. You need to solder pins on the board.

Olepir, 2019-09-15:
> Ravencore. I have 2 magic home rgb led controllers flashed with ravencore. After flash with ravencore you can’t controll it with normal ir controller, only with phone/tablet. You need to solder pins on the board.

Are you sure that Ravencore is compatible with single channel controllers?

mafyata, 2019-09-15:
Sorry , I don’t know for 1 channel. I have rgb.

curla92, 2019-09-16:
> How to flash Magic Home Single Channel with RavenCore or HAA?
> Can I flash it with RavenCore? If not – how to use HAA and is there template available for this LED controller?
> Thanks

https://github.com/curla92/LED-Dimmer-Homekit

Olepir, 2019-09-17:
> https://github.com/curla92/LED-Dimmer-Homekit

Any tutorials hot to use it?

curla92, 2019-09-17:
> > https://github.com/curla92/LED-Dimmer-Homekit
> 
> Any tutorials hot to use it?

you have to install with ota method. After you have to set it into eve app

Olepir, 2019-09-17:
> you have to install with ota method. After you have to set it into eve app

Any links for this ota method? I only familiar with RavenCore flashing.

curla92, 2019-09-17:
> > you have to install with ota method. After you have to set it into eve app
> 
> Any links for this “ota method”? I only familiar with RavenCore flashing.

```
esptool.py -p /dev/<your_ESPPort> --baud 115200 write_flash -fs 1MB -fm dout -ff 40m \
0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin
```

OTA REPO

`curla92/LED-Dimmer-Homekit
`
as raven method.

Olepir, 2019-09-17:
> OTA REPO
> 
> `curla92/LED-Dimmer-Homekit `

So everything is the same except OTA Repo?

curla92, 2019-09-17:
> > OTA REPO
> > `curla92/LED-Dimmer-Homekit `
> 
> So everything is the same except OTA Repo?

yes.

Olepir, 2019-09-17:
Thanks! Will try it now.

Olepir, 2019-09-17:
> yes.

Standard RavenCore pin is not accepted. Any ideas?

curla92, 2019-09-17:
[image]

Olepir, 2019-09-17:
I don’t see it in Eve because I can’t add it to HomeKit. RavenCore pin number is incorrect.

curla92, 2019-09-17:
> I don’t see it in Eve because I can’t add it to HomeKit. RavenCore pin number is incorrect.

code is 111-111-111

Olepir, 2019-09-17:
Thanks, working now, but ON is OFF, brightness minus is plus and vice versa. Polarity is ok.
Also how can I change wi-fi network?

curla92, 2019-09-17:
Check eve config. Please reebot after set. What kind of magichome do you have?

Olepir, 2019-09-17:
Working now, don’t know what I did… Is there any way to reset wifi setting for reconnect to different wiffi?

bj-tracer, 2019-09-30:
> Working now, don’t know what I did… Is there any way to reset wifi setting for reconnect to different wifi?

"Within first minute since device boots, wait about 5 seconds to allow device to boot completely, and then toggle any switch quickly (from Home App or from external switches) at least 10 times."
It's from wiki fo RavenCore firmware.
But you haven't any button... Use EVE for wi-fi reseting :)

Olepir, 2019-09-30:
> "Within first minute since device boots, wait about 5 seconds to allow device to boot completely, and then toggle any switch quickly (from Home App or from external switches) at least 10 times." [...]

Firmware from curla92 doesn’t have it.

RavenSystem, 2019-09-30:
I think if you have questions about a firmware outside of this project, you should ask them in the corresponding repo, not here.

closed 2019-09-30 by RavenSystem