RavenSystem/esp-homekit-devices #504: RGB W1 W2 LED Strips and Controller H801

magichew, 2019-09-18:
Hi all

I have the H801 LED controller. https://github.com/arendst/Sonoff-Tasmota/wiki/H801

I have it working fairly satisfactorily using the RavenCore using device type 14 and I've address the GPIOs to the correct LEDs. 

My LED strip has 5 colours RGB W1 & W2 but can only utilise RGB and W1 or W2 using this firmware. Is there a way to address the other white colour?

RavenSystem, 2019-09-19:
Sorry, actually only one white channel is supported.

closed 2019-09-19 by RavenSystem

magichew, 2019-10-24:
Sorry to bring this back but...
would it be possible to achieve via Home Accessory Architect? I haven't used that as it seems a little less user friendly as it needs a little more set up but I'd be prepared to investigate further if it could be done.

RavenSystem, 2019-10-24:
HAA does not support it yet. This feature will be included in the future.