RavenSystem/esp-homekit-devices #506: Nothing happens when setting up Wifi and OTA repository/binary

JHurk, 2019-09-21:
Hi,

I have followed all the steps as per your installation guide.
On step 5 I click 'Join' after filling all the details needed (SSID, password, repo and binary).
After waiting for 7 (and I have even waited more then 15 minutes at one time) nothing happens. The flash was successful but initial setup is not working. When clicking Join I do get a waiting screen, so the command 'seems to work'.

The switch (TH16) stays in AP-mode and the temporary SSID (LCM-XXXXXX) is always available. And when I connect to the SSID (LCM-XXXXXX) after 7 minutes I get the setup screen again. So it really looks as nothing happened after clicking Join.

I have flashed (and erased_flash) a few times and flashed through terminal and with a Mac-app called ESP8266Flash, both ways do not work. Flashing the Tasmota firmware through Atom does work. But want to use the switch within Homekit instead of Domoticz.

Any ideas?

JHurk, 2019-09-21:
I erased the flash and ran the installation without OTA, this works fine. So I got it connected to Homekit and it runs like a charm. Sorry to sound like a noob, but the only thing I am missing now is OTA updates of the firmware? Correct? Would be nice to have that, so would eventually like to have the OTA-version installed.

RavenSystem, 2019-09-28:
Logs, please.

myxa27, 2019-09-29:
Hey.  I have the same problem! Shelly RHBA

myxa27, 2019-09-30:
I tried to flash without OTA, the same thing...

HomeACcessoryKid, 2019-10-01:
Hi all, I'm HacK and the author of the LCM = OTA library

There are some - so far unexplained - moments that OTA doesn't want to cooperate with GitHub. I've been there myself feeling desperate for like 5 hours.
But then, without having changed anything, it just works and never complains again.

So, please make logs and lets see if they show new stuff or just the normal stalling.

logs can be made by running in macOS this command: `nc -kulnw0 45678`

myxa27, 2019-10-03:
Once again I requested and it all worked. Changed nothing. Below is the log ...

`
user-init-start
user-init-done
wifiready-done
--- ota_boot...0
OTABOOT VERSION: 1.0.0
--- ota_init
userbeta='0' otabeta='0'
active_sector: 0x0
--- ota_set_verify...OFF
--- DNS: done!
--- ota_get_version
--- ota_connect LocalPort=f09a DNS IP:140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..

user-init-start
user-init-done
wifiready-done
--- ota_boot...0
OTABOOT VERSION: 1.0.0
--- ota_init
userbeta='0' otabeta='0'
active_sector: 0x0
--- ota_set_verify...OFF
--- DNS: done!
--- ota_get_version
--- ota_connect LocalPort=cb37 DNS IP:140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
GET /HomeACcessoryKid/life-cycle-manager/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for --- ota_boot...0
--- ota_compare 1.0.0 with 1.0.0=0
HomeACcessoryKid/life-cycle-manager@version:"1.0.0"
--- ota_get_file
--- ota_get_file_ex
GET /HomeACcessoryKid/life-cycle-manager/releases/download/1.0.0/certs.sector HTTP/1.1
Host: github.com

--- ota_connect LocalPort=cb38 DNS IP:140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 302
--- ota_connect LocalPort=cb39 DNS IP:52.216.82.0 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
erasing@0xfa000> collected 4096 bytes
--- ota_finalize_file
active_cert_sector: 0xfa000
--- ota_get_pubkey
ret: 0
--- ota_get_privkey
--- ota_boot...0
--- ota_write_status
--- ota_load_user_app
user_repo='RavenSystem/ravencore' user_version='0.0.0' user_file='main.bin'
--- ota_boot...0
--- ota_get_version
--- ota_connect LocalPort=cb3a DNS IP:140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
GET /RavenSystem/ravencore/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for --- ota_boot...0
--- ota_compare 1.1.1 with 1.0.0=1
RavenSystem/ravencore@version:"1.1.1"
--- entering the loop
--- ota_get_pubkey
ret: 0
--- ota_set_verify...OFF
--- ota_get_version
--- ota_connect LocalPort=cb3b DNS IP:140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
GET /HomeACcessoryKid/life-cycle-manager/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for --- ota_boot...0
--- ota_compare 1.0.0 with 1.0.0=0
HomeACcessoryKid/life-cycle-manager@version:"1.0.0"
--- ota_get_hash
--- ota_get_file_ex
GET /HomeACcessoryKid/life-cycle-manager/releases/download/1.0.0/certs.sector.sig HTTP/1.1
Host: github.com

--- ota_connect LocalPort=cb3c DNS IP:140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 302
--- ota_connect LocalPort=cb3d DNS IP:52.216.145.131 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
 collected 155 bytes
--- ota_verify_hash
--- ota_hash
--- ota_set_verify...ON
certs size: 2628
TIME: Thu Oct  3 10:46:44 2019
--- ota_get_version
--- ota_connect LocalPort=cb3e DNS IP:140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
GET /HomeACcessoryKid/life-cycle-manager/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for --- ota_boot...0
--- ota_compare 1.0.0 with 1.0.0=0
HomeACcessoryKid/life-cycle-manager@version:"1.0.0"
--- ota_get_hash
--- ota_get_file_ex
GET /HomeACcessoryKid/life-cycle-manager/releases/download/1.0.0/certs.sector.sig HTTP/1.1
Host: github.com

--- ota_connect LocalPort=cb3f DNS IP:140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 302
--- ota_connect LocalPort=cb40 DNS IP:52.216.129.147 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
 collected 155 bytes
--- ota_boot...0
--- ota_get_hash
--- ota_get_file_ex
GET /HomeACcessoryKid/life-cycle-manager/releases/download/1.0.0/otaboot.bin.sig HTTP/1.1
Host: github.com

--- ota_connect LocalPort=cb41 DNS IP:140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 302
--- ota_connect LocalPort=cb42 DNS IP:52.216.165.251 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
 collected 155 bytes
--- ota_get_version
--- ota_connect LocalPort=cb43 DNS IP:140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
GET /HomeACcessoryKid/life-cycle-manager/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for --- ota_boot...0
--- ota_compare 1.0.0 with 1.0.0=0
HomeACcessoryKid/life-cycle-manager@version:"1.0.0"
--- ota_get_hash
--- ota_get_file_ex
GET /HomeACcessoryKid/life-cycle-manager/releases/download/1.0.0/otamain.bin.sig HTTP/1.1
Host: github.com

--- ota_connect LocalPort=cb44 DNS IP:140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 302
--- ota_connect LocalPort=cb45 DNS IP:52.217.36.100 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
 collected 155 bytes
--- ota_verify_hash
--- ota_hash
--- ota_hash
--- ota_get_file
--- ota_get_file_ex
GET /HomeACcessoryKid/life-cycle-manager/releases/download/1.0.0/otamain.bin HTTP/1.1
Host: github.com

--- ota_connect LocalPort=cb46 DNS IP:140.82.118.3 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 302
--- ota_connect LocalPort=cb47 DNS IP:52.216.241.52 local..OK remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..OK
erasing@0x9a000>^Collected 53248 bytes
`

joseluislucio, 2019-10-03:
> Hi all, I'm HacK and the author of the LCM = OTA library
>
> There are some - so far unexplained - moments that OTA doesn't want to cooperate with GitHub. I've been there myself feeling desperate for [...]

Whats the command to get the log in windows? I use esptool to flash it.

closed 2019-11-07 by RavenSystem