RavenSystem/esp-homekit-devices #512: Sonoff 4ch

lfiocco, 2019-09-26:
Sonoff 4ch, under ios13, everything appears as one device . Same is true for temp+switch. Can we split this up?

mafyata, 2019-09-26:
It is same with th10/16(device type 6).

KevinMiddleton, 2019-09-26:
You have to use the HAA firmware to get around this right now.  My JSON for Sonoff 4CH R2 is:

{
   "c":{
      "l":13,
      "i":1
   },
   "a":[
      {"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0 ,"p":0},{"g": 0,"t":1}]},
      {"0":{"r":[{"g":5, "v":0}]},"1":{"r":[{"g":5 ,"v":1}]},"b":[{"g":9 ,"p":0},{"g": 9,"t":1}]},
      {"0":{"r":[{"g":4, "v":0}]},"1":{"r":[{"g":4 ,"v":1}]},"b":[{"g":10,"p":0},{"g":10,"t":1}]},
      {"0":{"r":[{"g":15,"v":0}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14,"p":0},{"g":14,"t":1}]}
   ]
}

lfiocco, 2019-09-27:
I have not used HAA, can I install it on an a th16? Can it show temp?

RavenSystem, 2019-09-28:
Improved Sonoff 4CH JSON:
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#sonoff-4ch-it-creates4-homekit-switches-and-gpio-0-button-can-be-used-to-enter-in-setup-mode-when-you-press-it-and-hold-more-that-10-seconds-for-the-first-30-seconds-since-boot

closed 2019-09-28 by RavenSystem

Partizan7676, 2019-09-29:
Try latest template for 4CH Pro rev2 - not work, template from KevinMiddleton - device works, but can not add it to home - nothing to add Home app - nothing appear when i try to add accessory.

KevinMiddleton, 2019-09-29:
I have used the template with two 4CH R2 devices with complete success.  The devices need to be on the same network as the home app to be found.

Partizan7676, 2019-09-29:
I know that. Thanks for your comment, but at this moment with your template device begin works and Home app cannot see it :( with RavenCore - works fine.
BTW How with your template enter to Setup Mode ?

KevinMiddleton, 2019-09-29:
I believe it is the standard - press L1 (button0) 10 times or more in the first 30 seconds after boot.

Not 100% sure why RavenCore works but HAA isn't working for you.

Partizan7676, 2019-09-29:
Alredy done. With 10 time press, also reset HomeKit ID - still device doesnt appear in HomeApp

KevinMiddleton, 2019-09-29:
what if you install RavenCore, use EVE (setup section for device) to push the HAA via OTA?  I did that for at least one of my devices as a test.

Partizan7676, 2019-09-29:
Exactly do that like you, but result is same.

KevinMiddleton, 2019-10-03:
apparently in iOS 13.2, there will be an option to split the device so it should appear as 4 again using RavenCore (saw this info on https://www.reddit.com/r/HomeKit/comments/dceaqd/it_worked_apple_finally_listened_to_us_and/)

lfiocco, 2019-10-03:
Ya, saw that. It’s really messing up my camera setup. I have a fix now. Might keep it. 

Thanks for your help