RavenSystem/esp-homekit-devices #513: RavenCore not saved Init State TH in heater mode

akaGDM, 2019-09-29:
RavenCore  Problem with the Sonoff basic and TH-10 Thermostat (Device type 5).  In the modes type 1 (heater) and type 2 (cooler) after rebooting, the status “Init state TH” is not saved.  The device boots off.  In type 0 mode, everything works correctly, the status is saved.  Can this be fixed?

vsidlu, 2019-10-02:
I also noticed that in type 1 (heater) thermostat is somehow switching off. Also changing "init state TH" is not helping. There where no problems in ios12, but in ios13 thermostat looks different and seems also  behaves different. My conclusion is  - when you turn on thermostat and set target temp. after it reaches target temp. it turns off. Previously after reaching target temp. it just changed mode from heating to off and remained turned on, so after temp. drop bellow target temp. it changes mode to heating. For now switching to type 0 mode is answer. As there is HAA in progress do not know if there will be changes in RavenCore. Will see!

RavenSystem, 2019-10-03:
Last state is saved when 5 seconds pasts after last change. Did you considerer that?

claire078, 2019-10-07:
I can confirm both problems with mode thermostat.
I don’t really understand that you asked.

RavenSystem, 2019-11-03:
RC is deprecated.

closed 2019-11-03 by RavenSystem