RavenSystem/esp-homekit-devices #514: Best way to get IR and how to .json in HAA

lfiocco, 2019-09-29:
What’s he best way to get IR sensor into HomeKit with sonoff and RavenCore?

How do I change and load json file into HAA

mgrapiniak, 2019-10-13:
> How do I change and load json file into HAA

I can answer the second part of your question. 
If you already have an accessory with HAA onboard you can just toggle the switch state from Home app or by using physical switch (if avail) for like 10 or more times... then just wait... accessory will enter Setup mode... then you can use your phone and go to WiFi settings... choose HAA-xxxx from the list and then follow up instructions from: 
https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect#setup-mode

This should help :)

RavenSystem, 2019-10-13:
IR is nor supported yet.

closed 2019-11-07 by RavenSystem