RavenSystem/esp-homekit-devices #515: HAA Inching question

caligari74, 2019-09-29:
First of all, thank you very very much for the fantastic software!

Im having a problem figuring out the right syntax for getting inching to work with HAA.
I'm using the following JSON to create 2 accessories in HomeKit, and the HW is a D1 mini.

What I want is for two relays to remain off after boot.  I want the HomeKit switches to remain off, until they are clicked once turn on and then automatically shut off after 0.5 seconds. The version of the JSON is below.

Right now the (erroneous) behavior is that both relays are immediately turned on after boot, like the switch is inverted.  When I move the "i": 0.5 from the "1" to the "0" activity, the relays remain off as expected.  HomeKit shows the switch as off.  I turn the switch on in HomeKit, and the relay remains off.  I then turn off the switch in HomeKit, and then the relay is triggered for .05 seconds--it turns on and then off.

Am I doing something wrong, or is this a bug?

Thanks again!

=====

{
  "c": {
    "o": 1,
    "l": 13
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12,
            "v": 0
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1,
            "i": 0.5
          }
        ]
      }
    },
    {
      "0": {
        "r": [
          {
            "g": 5,
            "v": 0
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 5,
            "v": 1,
            "i": 0.5
          }
        ]
      },
      "b": [
        {
          "g": 0
        },
        {
          "g": 14
        },
        {
          "g": 14,
          "t": 0
        }
      ]
    }
  ]
}

caligari74, 2019-09-30:
I was able to get the inching to work with this:

{"c":{"o":0},"a":[{"0":{"r":[{"g":12,"v":1}]},"1":{"r":[{"g":12,"v":0,"i":0.5}]}},{"0":{"r":[{"g":5,"v":1}]},"1":{"r":[{"g":5,"v":0,"i":0.5}]},"b":[{"g":0},{"g":14},{"g":14,"t":0}]}]}

Now the after boot the relays stay off, and HomeKit shows they are off.  Clicking a button in the app to turn a relay on actually does turn it on, and for the 0.5 seconds. The relay then clicks off, but the button remains "ON" in the Home app.

RavenSystem, 2019-09-30:
I think that you want inching to work on the accessory, and not on the relay. Remove “i” from action relay arrays, and set “i” as a key of accessory, like “0” and “1”.

caligari74, 2019-09-30:
Thanks for the tip.  This is now working:

{"c":{"o":0},"a":[{"0":{"r":[{"g":12,"v":1}]},"1":{"r":[{"g":12,"v":0}]},"i":0.5},{"0":{"r":[{"g":5,"v":1}]},"1":{"r":[{"g":5,"v":0}]},"i":0.5,"b":[{"g":0},{"g":14},{"g":14,"t":0}]}]}

closed 2019-10-01 by RavenSystem