RavenSystem/esp-homekit-devices #516: Any HAA tutorial for 3gang Switch with status led ?

Codkit, 2019-09-29:
Hi all

I have a 3gang wall switch ,there are 3 leds corresponding to each switch, how should I add the “c” and “l” data ? Also any other parameters for the “l” like on and off, right now the led on my Switch only blink once whenever I press it, it won’t stay on. Thanks all for your help.

bj-tracer, 2019-09-30:
Hi!
I think you may use for it relay declaration at relay array.
"c" using for led indicated status of esp. You no need to use it for your case.

Codkit, 2019-10-01:
Hi bj-tracer

Thanks for your help , I put led’s gpio with “g” in relay array , it seems ok
But the leds status is reversed whenever reboot is needed.

Codkit, 2019-10-01:
Hi all
gpio13 = relay
gpio16 = Switch led
{"c":{"l":0,"i":1},"a":[{"0":{"r":[{"g":13,"v":0},{"g":16,"v":1}]},"1":{"r":[{"g":13,"v":1},{"g":16,"v":0}]},"b":[{"g":12}]}]}
Appreciate if anyone can help me figure how to reverse the led state when reboot, should I use “s” and where to put it ? Right now after reboot relay is off but led is on.
Thanks everyone

Codkit, 2019-10-01:
Seems ok after updated to 0.1.1, leds initial state is off (reversed).
I’d like to apply “s”:5 (last state), can anyone show me an example of how to apply “s” with last state ? I keep getting error....... thanks very much

pilot1981, 2019-10-01:
Initial state 5 last state

{"c":{"l":0,"i":1},"a":[{"s":5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0},{"g":13,"t":0}]},{"s":5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]}]}

Codkit, 2019-10-01:
Hi pilot1981

Thanks you so much for your quick reply !

pilot1981, 2019-10-01:
Say thanks to @RavenSystem !!!

Codkit, 2019-10-01:
Yes , big thanks to José , amazing work ! Few of my old wall switches are back to life !

RavenSystem, 2019-10-01:
You are welcome. But users' feedbacks and community helps are very important too!

closed 2019-10-01 by RavenSystem