RavenSystem/esp-homekit-devices #517: Problem with Sonoff Touch T1 1 channel

LukasNowak, 2019-09-30:
Dear's 
I install the software correctly in Sonoff Touch T1, after I connect them to the power I can't find the WiFi network of the device, WiFi LED is off, also when I click the main button nothing happened, when I click the reset button the backlight of the main button goes out, after releasing the button it lights up again. Can you help me?

RavenSystem, 2019-10-13:
What firmware did you installed? RC or HAA? With or without OTA? We need more details of your steps.

LukasNowak, 2019-10-14:
RC with OTA

Z wyrazami szacunku/Best regards

Łukasz Nowak

LukasNowak, 2019-10-24:
Can I ask about short manual how to install RavenCore in Sonoff T1 1 channel? Which version?

RavenSystem, 2019-11-03:
RC is deprecated.

closed 2019-11-03 by RavenSystem