RavenSystem/esp-homekit-devices #519: Devices mostly unreachable since iOS 13

CD650, 2019-10-02:
Since iOS 13 the devices are mostly unreachable.

RavenCore: 1.1.1 (mostly sonoff basic)
iOS on iPad (server): 13.1.2
iOS on iPhone: 13.1.2

For example I have an automation for a motion sensor which triggers devices with ravencore and non-ravencore devices. 8-9 times of 10 only the other devices are turned on/off. It seems that if the home app is not freshly opened on any devices after that the devices will be unavailable.

It was working before iOS 13 well.

I updated to 1.1.1 almost at the same time with iOS 13.

jat80, 2019-10-05:
I have a sonoff SV (garage door) and having similar issues.

St0nehead, 2019-10-06:
Same problem here ☹️

Fernyman79, 2019-10-12:
Same thing here. I have two sonoff basic that were rock solid previously. I can see that they are stably connected to my network via my router UI (unifi) but are not reporting to HomeKit.

nasawz, 2019-10-29:
Same thing here.

SmartArduino, 2019-10-29:
Maybe mDNS doesn't have enough memory，Whether the restart device is normal

CD650, 2019-10-29:
It would be great to solve this issue, even the RavenCore is not developed, because I was satisfied with it, until it was working well.

jat80, 2019-10-29:
My garage door accessory is actually fine now.  I haven’t had any problems with iOS 13.1.3 and iOS 13.2.

SmartArduino, 2019-10-30:
ok, that's a great idea.

With the best regards
Yichone (Manager)
DOIT Ltd., from Shenzhen

Resources and documents: wiki.doit.am
site: www.doit.am 
shop: www.vvdoit.com, www.smartarduino.com 
Skype: yichone
whatsapp: +8618676662425
wechat: 18676662425
Emails: yichone@doit.am, yichoneyi@163.com 
 
From: JT
Date: 2019-10-30 05:37
To: RavenSystem/esp-homekit-devices
CC: yichone; Comment
Subject: Re: [RavenSystem/esp-homekit-devices] Devices mostly unreachable since iOS 13 (#519)
My garage door accessory is actually fine now. I haven’t had any problems with iOS 13.1.3 and iOS 13.2.

SmartArduino, 2019-10-30:
now, your homekit is working normally?

With the best regards
Yichone (Manager)
DOIT Ltd., from Shenzhen

Resources and documents: wiki.doit.am
site: www.doit.am 
shop: www.vvdoit.com, www.smartarduino.com 
Skype: yichone
whatsapp: +8618676662425
wechat: 18676662425
Emails: yichone@doit.am, yichoneyi@163.com 
 
From: CD650
Date: 2019-10-29 21:51
To: RavenSystem/esp-homekit-devices
CC: yichone; Comment
Subject: Re: [RavenSystem/esp-homekit-devices] Devices mostly unreachable since iOS 13 (#519)
It would be great to solve this issue, even the RavenCore is not developed, because I was satisfied with it, until it was working well.

CD650, 2019-10-31:
I had configured an iPad with iOS 12 as a HUB (instead of iOS 13) and with 12 everything is working fine.

Is it an iOS issue, or something needs to be updated in the firmware to work along with iOS 13?

CD650, 2019-10-31:
The automations are not working under iOS 13.

For example, I have a Aqara motion sensor in my wardrobe. When I open it and it detects motion, it turns on a Sonoff Basic lamp with  RavenCore 1.2

It is working fine under iOS 12.

But with iOS 13, I just see that the motion is detected but the light is not turned on only 8 times / 10 times. But if I wake up the iPad by opening the screen it will work.

Under iOS 12 it is always working.

RavenSystem, 2019-11-03:
I don't have any issues with HomeKit. I use 2 Netatmo cameras, 3 Koogeek lights, and many, many, many HAA devices.

closed 2019-11-07 by RavenSystem