RavenSystem/esp-homekit-devices #520: Sonoff 4CH PRO R2 unreachable after JSON config

milonet, 2019-10-02:
Hi, after first setup with OTA, when i put the JSON string, my sonoff 4CH pro R2 become unavailable. The relay make fast "tac tac" and the network becomes down

the json string is: 
`{"c":{"l":13,"i":1},"a":[{"i":0.5,"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0 ,"p":0},{"g": 0,"t":1}]},{"i":0.5,"0":{"r":[{"g":5, "v":0}]},"1":{"r":[{"g":5 ,"v":1}]},"b":[{"g":9 ,"p":0},{"g": 9,"t":1}]},{"i":0.5,"0":{"r":[{"g":4, "v":0}]},"1":{"r":[{"g":4 ,"v":1}]},"b":[{"g":10,"p":0},{"g":10,"t":1}]},{"i":0.5,"0":{"r":[{"g":15,"v":0}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14,"p":0},{"g":14,"t":1}]}]}`

It's correct?

milonet, 2019-10-02:
An update: If i put the sonoff in setup mode pressing 10 time a physical button, it connect to my wifi home. When i click on save to update wifi home settings, the sonoff becomes unavailable until put in setup mode. It seem that keep the wifi settings only in setup mode..

RavenSystem, 2019-10-13:
I don’t know. Try it and comment, please.

milonet, 2019-10-19:
Hi, i tried this string: `{
  "c": {
    "l": 13,
    "i": 1,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "t":4,"0": {
        "r": [
          {
            "g": 12,
			"i":0.4,
			"v":1
		
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12
          }
        ]
      },
      "b": [
        {
          "g": 0
        }
      ]
    },
    {
      "t":4,"0": {
        "r": [
          {
            "g": 5,
			"i":0.4,
			"v":1
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 5
          }
        ]
      },
      "b": [
        {
          "g": 9
        }
      ]
    },
    {
      "0": {
        "r": [
          {
            "g": 4
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 4,
            "v": 1
          }
        ]
      },
      "b": [
        {
          "g": 10
        }
      ]
    },
    {
      "0": {
        "r": [
          {
            "g": 15
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 15,
            "v": 1
          }
        ]
      },
      "b": [
        {
          "g": 14
        }
      ]
    }
  ]
}` 
do you confirm that is good?
I have two problem:
1) When device power on the relay on button 0 inching .. this is not good because i will connect the relay to lock mechanism
2) the devices exit from wifi. the only way to reach it on HOME (non captive portale) wifi network is putting on setup mode. After save the string

RavenSystem, 2019-10-19:
1. When device power on, the relay in GPIO 12 should be low (value:0, off...). Please, include "o":1 in config section and send the log. I don't see any relay activation at boot; all relays are off.

2. Yes, the issue is fixed with the newest version 0.3.3 I have released just now. (Tested with your JSON).

milonet, 2019-10-19:
Hi, thak you for you support. Much appreciated.
The wifi issue is fixed, the device still reachable after JSON uploaded.
But anoter problem is born. It's impossible to enter in setup mode. I press for 10-11 seconds the button 0 after device start (the led blink correctly) but i cannot enter in web page.
In your test does it work correctly?

milonet, 2019-10-19:
...and... how I can see the boot logs?

milonet, 2019-10-20:
Hi, i tested with success this JSON string: `{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":4,"i":0.4,"0":{"r":[{"g":12,"v":1}]},"1":{"r":[{"g":12,"v":0}]},"b":[{"g":0}]},{"t":4,"i":0.4,"0":{"r":[{"g":5,"v":1}]},"1":{"r":[{"g":5,"v":0}]},"b":[{"g":9}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}]},{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}]}]}`

The problem of initial inching on boot devices seems solved. WiFi after JSON setting is ok.
For me 0.3.3 is stable !!

closed 2019-10-20 by RavenSystem