RavenSystem/esp-homekit-devices #523: Reset WiFi option in HAA

mientki, 2019-10-04:
Can you add the "reset wifi settings" option in the next release version. This option will shorten the time the HAA-xxxxx network appears, when the known one is no longer available. This is the only way to change the network for another one at the moment. I do not want to wait 10 minutes, I would like to force the reset of wifi settings.

RavenSystem, 2019-10-04:
That is not make sense. You can change your wifi settings entering into setup mode. And the 10 minutes timeout was deprecated and removed some time ago for security reasons.

mientki, 2019-10-04:
OK. I had preview version which it has not been on yet. Now it works as it should. Configuration mode also introduces AP mode. Thank you. This soft is amazing. :-)

closed 2019-10-07 by RavenSystem