RavenSystem/esp-homekit-devices #524: HAA Garage Device

noobydp, 2019-10-05:
Hi José,
Thanks for all your hard work on HAA.

Would it be possible to have the garage controller added to HAA?
I know its available in ravencore already but I'd rather use HAA.

Thanks,
David

tunip, 2019-10-05:
I am in the procedure to flash all my devices with HAA. Garagedoor (Sonoff SV) is the only device which i can not flash with HAA.
Would be awesoume if garagedoor type would find its way into HAA.

Thx Joseph

RavenSystem, 2019-10-05:
Don't worry. Garage door will be in HAA.

perezjosue, 2019-10-22:
I am in the process of starting off with HAA. Would you recommend I just go ahead with RavenCore for the garage for now or will it be added to HAA soon?

RavenSystem, 2019-10-22:
If RavenCore garage door config is good for you, use it. If you want, then you can migrate to HAA without reflashing.

perezjosue, 2019-10-29:
> If RavenCore garage door config is good for you, use it. If you want, then you can migrate to HAA without reflashing.

Thank you for your reply and for creating all this.

closed 2020-01-10 by RavenSystem