RavenSystem/esp-homekit-devices #526: HAA - setup mode with hold press

macjl, 2019-10-06:
Hello,

In the general config Json, I’ve added « "b":[{"g":0,"t":5}] »
But I’m not able to enter in setup mode. How does the « hold press» works? How long should I press the button? When?

RavenSystem, 2019-10-06:
What HAA version are you using?

macjl, 2019-10-06:
Latest : 0.2

RavenSystem, 2019-10-06:
You must hold button 11 seconds, before 5 seconds since device boots.

macjl, 2019-10-06:
It does not seems to work.
Here is my complete JSON : 
`{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0,"t":0}]}]}`

RavenSystem, 2019-10-06:
Change t:0 to t:1 (or remove it because it is the default).

macjl, 2019-10-06:
Thanks for the answer!
By setting t:1, it's working. But I would prefer to keep t:0, because it's much more responsive to switch the relay. Is there a way to do that?

RavenSystem, 2019-10-06:
No, sorry. t:0 is incompatible with t:5.

RavenSystem, 2019-10-06:
But you can remove hold press option, and enter setup mode pushing button 10 times quickly.

macjl, 2019-10-06:
I wil use that for the moment. But I'v seem that 'pushing 10 times quickly' is not very reliable. For example, I was not able to use it with device type 12 (movement sensor).

RavenSystem, 2019-10-06:
Toggle quickly doesn’t work with sensor accessory types. Best way should be to use a different GPIO to enter setup mode.

macjl, 2019-10-06:
Ok, I understand the logic.
Is there a plan to use an HomeKit accessory as a trigger for setup mode? It would be easier to enter in setup mode, as it was with RavenCore.

RavenSystem, 2019-10-06:
You can add a dummy switch.

macjl, 2019-10-06:
Of course! I forgot it. Thanks!

closed 2019-10-06 by macjl

claire078, 2019-12-30:
“General: Now device in setup mode will reboot in normal mode automatically after 90 seconds if setup web page is not loaded and Auto OTA Updates is disabled.”

Would be start firmware with this after power cut?
Or it could be a security risk?