RavenSystem/esp-homekit-devices #528: motion sensor type 5 for sonoff basic

vsidlu, 2019-10-06:
would like to add motion sensor to sonoff basic gpio 14. can not figure out right json for sensor to work. was able to add motion accessory but thats all. need json sample for type 5. thanks in advance.

bj-tracer, 2019-10-07:
I guess it seems like a  "a":[{"t":5,"0":{"f0":[{"g":14}]},"1":{"f1":[{"g":14}]}}]
I think it need for inverting state of HK sensor and real buttons. ->"a":[{"t":5,"0":{"**f1**":[{"g":14}]},"1":{"**f0**":[{"g":14}]}}]

RavenSystem, 2019-10-07:
@bj-tracer Sorry, you are wrong. "f0" and "f1" can not be into "0" and "1". They are different things.

A motion sensor accessory could be:
{"t":5,"f0":[{"g":14,"t":0}],"f1":[{"g":14}]}

Or if you want a better approach, you can use inching to keep the HomeKit motion sensor activated for 30 seconds, for example:
{"t":5,"i":30,"f1":[{"g":14}]}
See that "f0" is removed because when "f1" is launched, after 30 seconds the HomeKit motion sensor state will be 0 again automatically, without any external interaction.

An additional option could be adding hardware actions when an event occurs, so here is where "0" and "1" can be used. For example, when sensor detects movement, trigger internal relay to turn on a siren for 10 seconds:
{"t":5,"i":30,"f1":[{"g":14}],"1":[{"g":12,"v":1,"i":10}]}

mafyata, 2019-10-07:
How did you do the sonoff basic to have motion sensor? Is it hardware connected to gpio pins?

vsidlu, 2019-10-07:
@RavenSystem thanks for explanation and samples. I will use sonoff basic with motion sensor to turn on outside lights when there is movement (and send snapshot from camera). So last sample is not for this situation as it will turn on lights also in daytime. So I will use second json and add automation in home app to work only nighttime. Thanks for your effort!

closed 2019-10-07 by vsidlu

vsidlu, 2019-10-07:
> How did you do the sonoff basic to have motion sensor? Is it hardware connected to gpio pins?

yes! you connect PIR sensor to gpio 14. my PIR sensor also needed 5v, so additional soldering was necessary.

mafyata, 2019-10-07:
Can you post the link with instruction?

vsidlu, 2019-10-07:
> Can you post the link with instruction?

i still need to check if json is working and then i can post how i managed it!

bj-tracer, 2019-10-07:
@RavenSystem Thanks for reply. I just supposed ;)

gerardwr, 2019-10-07:
@RavenSystem said:
> Or if you want a better approach, you can use inching to keep the HomeKit motion sensor activated for 30 seconds, for example:
> `{"t":5,"i":30,"f1":[{"g":14}]}`

Had quite some trouble composing this in a valid JSON for HAA, but in the end I succeeded using:
```
{"a":[{"t":12,"i":10,"f1":[{"g":14}]}]}
```
Note that:
- I used "t":12 for a MotionSensor, "t":5 is for a ContactSensor
- inching for 10 secs i.o. 30 secs

If you check the Serial out it says:
```
15:47:36.141 -> HAA > ACCESSORY 0
15:47:36.141 -> HAA > Accessory type=12
15:47:36.141 -> HAA > Enable button GPIO 14, type=1, inverted=0
15:47:36.141 -> HAA > Sensor bool activated
```
The MotionsSensor is recognised by the IOS Home App, and changes status to  "MotionDetected" when the signal on GPIO14 changes from LOW to HIGH. After 10 seconds the status in the App is changed to the default state, ready for the next Low-to-High transition on GPIO14.

In the IOS App you can configure the MotionSensor to Notify when Motion is detected, both by a NotificationBar and by a Sound.

vsidlu, 2019-10-11:
> > Can you post the link with instruction?
> 
> i still need to check if json is working and then i can post how i managed it!

As promised link for Sonoff basic with motion sensor - [https://www.ljcaesar.com/2018/02/sonoff-smart-wifi-switches-tasmota-external-sensor-5v-pir-mod/](url)

Instead of Tasmota use HAA with json bellow - 
{"c":{"o":1,"l":13,"i":1},"a":[{"t":1,"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0,"p":1,"t":1}]},{"t":12,"i":10,"f1":[{"g":14}]}]}

all thanks to @RavenSystem

RavenSystem, 2019-10-11:
If you want, you can remove default values and UART log output, and add a button to enter setup mode (holding down to 10 seconds):

{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":["0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":12,"i":10,"f1":[{"g":14}]}]}

vsidlu, 2019-10-11:
thanks! you where missing { in accessory - see if I made right correction - 
{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":12,"i":10,"f1":[{"g":14}]}]}

RavenSystem, 2019-10-11:
Sorry. Yes, you are right.

vsidlu, 2019-10-11:
I also do not see accessory type 1. This could be right - 
{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":1,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":12,"i":10,"f1":[{"g":14}]}]}

How accessories are proceed to homekit? What happens if I change order - first t:12 and then t:1? Right now I see in homekit switch with accessory (motion). Is there way to add two separate accessories?

RavenSystem, 2019-10-11:
Accessory type 1 is the default option and it is not necessary.

If there are 4 accessories or less, first accessory works as accessory bridge. All accessories are individuals, because each service uses its own accessory.

vsidlu, 2019-10-11:
thanks! now it is clear for me!

mafyata, 2019-10-11:
So with motion it will on/off , but can I switch to ON manually from home app? And second question- if I switch to ON from home app and detect motion will it switch to off after XX seconds by motion sensor?

vsidlu, 2019-10-12:
> So with motion it will on/off , but can I switch to ON manually from home app? And second question- if I switch to ON from home app and detect motion will it switch to off after XX seconds by motion sensor?

if you use this json - {"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":1,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":12,"i":10,"f1":[{"g":14}]}]}

you will have two separate accessories - switch and motion sensor (not related to each other). you can make any automation using homekit automations. example i use (switch on for 10 minutes after there is no motion) - 
automation rule 1 - if there is motion turn on switch for 1 hour.
automation rule 2 - if there is no motion turn on switch for 10 min.

if you want to trigger switch with motion without any rule you need to use @RavenSystem explained json - 
An additional option could be adding hardware actions when an event occurs, so here is where "0" and "1" can be used. For example, when sensor detects movement, trigger internal relay to turn on a siren for 10 seconds:
{"t":5,"i":30,"f1":[{"g":14}],"1":[{"g":12,"v":1,"i":10}]}

mafyata, 2019-10-12:
Where did you make automations? In Shortcuts app?

vsidlu, 2019-10-12:
Apple Home app!

mafyata, 2019-10-12:
Did you see this:
http://techgurka.blogspot.com/2013/05/cheap-pyroelectric-infrared-pir-motion.html
I think that I don't need to solder 5v cable on my sonoff basic.

vsidlu, 2019-10-13:
yes i know this option. will check if it is reliable.

mafyata, 2019-10-15:
So , my sonoff basic is r2 and I can’t use gpio14. Do you know which gpio I can use instead of gpio14?

mafyata, 2019-10-15:
I found this on the net. 
[image]
[image]

RavenSystem, 2019-10-15:
You can use GPIO3.

mafyata, 2019-10-15:
Do you mean RX ?

RavenSystem, 2019-10-15:
GPIO3 is RX or TX, I can’t remember now.

bj-tracer, 2019-10-16:
GPIO3 = RX
And this board has GPIO2 instead GPIO14. But some sensors incorrect working with it.

mafyata, 2019-10-18:
Can I use sonoff th16 instead of sonoff basic. It will be cool that I can make a plug for the pir sensor. 
[image]

RavenSystem, 2019-10-18:
@mafyata You can use any device you want.

mafyata, 2019-10-19:
Sorry , I mean with this json for sonoff basic.

claire078, 2019-12-30:
> 
> 
> GPIO3 is RX or TX, I can’t remember now.

I'd like to use a motion accessory with sonoff R2 board.
It didn't work.
I tried simple switch with external switch on GPIO3 - RX, but it doesn't work too.

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":3},{"g":3,"t":0}]}]}

RX should with GND close, right?

mafyata, 2020-02-02:
Hey , I'm trying to set up Sonoff th16 with pir sensor on gpio14. On/Off work well , but pir is not detecting in home app.
My json is : {"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":12,"i":10,"f1":[{"g":14}]}]}
Is this json correct?

mafyata, 2020-02-02:
Now trying json from wiki : {"c":{"l":13},"a":[{"t":1,"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":12,"i":10,"f1":[{"g":14}]},{"0":{"r":[]},"1":{"r":[],"s":[{"a":1}]},"t":1,"b":[],"s":0}]}
Same result, no motion detection.