RavenSystem/esp-homekit-devices #529: Reducing power consumption

bj-tracer, 2019-10-07:
Hi, @RavenSystem !
Can you realize a power safe mode for esp8266 controllers?
It has 3 sleep modes - Modem-sleep, Light-sleep and Deep-sleep.
Must whanted mode is the Deep-sleep. Peoples say that will be reduce power consumption for 2500 mAh battery to 300 hours of ESP working.
This mode can be using at autonomus sensors.

bj-tracer, 2019-10-07:
I found same question at Max Kulkin's git.
It's not possible with HK :(

RavenSystem, 2019-10-09:
Yes, it is not possible because each HK accessory is a server that must be connected permanently to WiFi network to listen entry requests.

closed 2019-10-09 by RavenSystem

ericklil, 2020-04-08:
I would like to think `Deep-sleep` would be possible with the `Stateless Switch `sample. This unit does nor give error messages if left powered off as it appears there is only one way communication. It would be a simple way to make a remote control that can run on a `LiPo` battery.. Any chance of adding this?