RavenSystem/esp-homekit-devices #533: iOS notification support in HAA ?

mgrapiniak, 2019-10-12:
Hi,

I have flashed the Sonoff Basic switch with your latest HAA today and everything works great however I have noticed lack of Notification option in Home app as per Apple article here: 

https://support.apple.com/en-us/HT208939

Any chance to to implement such support as an enhancement request ?

Regards,
Marek

RavenSystem, 2019-10-13:
HAA supports notifications, like all HomeKit accessories, since that feature is Apple side. Nothing to do.

claire078, 2019-10-13:
> Hi,
> 
> I have flashed the Sonoff Basic switch with your latest HAA today and everything works great however I have noticed lack of Notification option in Home app as per Apple article here:
> 
> https://support.apple.com/en-us/HT208939
> 
> The problem is that without notifications I have no means to trace how many times switch was on or off and this feature would be quite useful.
> 
> Any chance to to implement notification support as an enhancement request ?
> 
> Regards,
> Marek

I think you can get notifications by locks and doors only. Security purposes.

RavenSystem, 2019-10-13:
Sensors have notifications too.

peros550, 2019-10-13:
This is where MQTT protocol would be a good Solution for tracking device status. A third party home automation system would be needed (HomeAssistant , openHab etc) and those systems could be used in parallel to Home app providing so much more! 

 MQTT is definitely  less secure  compared to HomeKit , but I guess this is always the question : usability vs security . Being in a closed LAN environment , someone may choose for more usability over security.

claire078, 2019-10-13:
> Sensors have notifications too.

Yes, you have right. I have forgotten the sensors. 👍

mgrapiniak, 2019-10-13:
I do not see notification for my sonoff basic switch in Home app ... I would attach photo but there is no option to attach.

RavenSystem, 2019-10-13:
@mgrapiniak What kind of HomeKit accessory do you have? Switches and outlets have not got notifications.

mgrapiniak, 2019-10-13:
> @mgrapiniak What kind of HomeKit accessory do you have? Switches and outlets have not got notifications.

Interesting... I didn’t know this. I do have a switch :(
Is it KomeKit thing or HAA limitation?

RavenSystem, 2019-10-13:
HomeKit limitation. Notifications are managed ONLY by Apple.

RavenSystem, 2019-10-13:
If you want notifications, you can change "t":1 with "t":4. You will have a lock mechanism icon, but your device will work like a switch and you will have notifications.

Locutus73, 2019-10-13:
> I think you can get notifications by locks and doors only. Security purposes.

Garage Doors and Security Systems (those with Stay, Away, Night and Off states) have notifications too. And, as other stated, sensors too.

Regards.

mgrapiniak, 2019-10-13:
> @mgrapiniak What kind of HomeKit accessory do you have? Switches and outlets have not got notifications.

Hey APPLE. This make no sense to me :) How can I then monitor status of switches ?!? 👎 I guess APPLE knows best what is good for the users 😄

mgrapiniak, 2019-10-13:
> If you want notifications, you can change "t":1 with "t":4. You will have a lock mechanism icon, but your device will work like a switch and you will have notifications.

That was very useful. Lock however use Apple like status so the logic is actually reversed. I wanted switch to be ON when the status is Unlock and I was able to figure it out by swapping "v" param for "0" and "1" state declaration.

Thanks for your help and others in this thread!

closed 2019-10-13 by mgrapiniak