RavenSystem/esp-homekit-devices #534: HAA S26 Unable to pair

brymonster, 2019-10-13:
First, thank you for all the wonderful tools developed here!
Total noob at all of this so I apologize in advance.

I believe I have successfully erased and flashed my Sonoff S26 with HAA. 
Soldering the contacts was much easier than expected.

[image]

Connected to the LCM-XXXX network, selected my home network, enter password.
OTA repository for Home Accessory Architect: RavenSystem/haa
OTA binary file: main.bin
 waited..
Tried pairing to HomeKit. Nothing found. Restarted device and phone. Same result, no accessories found. For a first time user, I followed the instructions to the "T", but it's still left unfinished. It seems there's still more to do.
At this time noticed the device has a HAA-XXXX SSID.

[image]

This is where I realized, I needed a JSON file to configure the device. 
For the S26, I'm using:
`{"c":{"l":13,"i":1},"h":[{"t":2,"a":{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]}},"b":[{"g":0}]}]}`

[image]

Unfortunately, despite multiple attempts, the device still won't pair with HomeKit. 
I have tried many times, unplugging, replugging and restarting iPhone/iPad.

[image]

[image]

Does my process seem right? 
If I missed something, please let me know. 
Thank you all again for the hard work in development and community support.

RavenSystem, 2019-10-13:
Your process seems right, but your JSON is wrong because it is old, from a beta version.

Try this from here:
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#sonoff-s20s26-without-uart-output-with-status-led-and-working-button

brymonster, 2019-10-13:
hey thanks!
I will try that now. 
Do I need to reconnect to my wifi every time I enter that page?
Anyway to know its successfully connected to my network?

closed 2019-10-13 by brymonster

RavenSystem, 2019-10-13:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect#setup-mode

RavenSystem, 2019-10-13:
> Do I need to reconnect to my wifi every time I enter that page?
No.

> Anyway to know its successfully connected to my network?
Only using your network devices (your router wifi web, dhcp server...).

brymonster, 2019-10-15:
thank you @RavenSystem! 
looking forward to iFan03 HAA support...and for the rest of my sonoffs to arrive.