RavenSystem/esp-homekit-devices #535: [HAA 0.2.3] LED status on Sonoff Basic switch

mgrapiniak, 2019-10-13:
Hi!,

I would like to achieve following:
- when the relay is ON the LED should show solid light
- when the relay is OFF the LED should show no light

Is it possible to get this setup in HAA 0.2.3 ?

RavenSystem, 2019-10-13:
Of course. Simply declare it into actions, because a LED is a digital output.

mgrapiniak, 2019-10-14:
> Of course. Simply declare it into actions, because a LED is a digital output, like a relay. And remove status LED from config section.

Shall I remove "l": 13 from "c" section and put in into accessory section where "g" and "v" is declared for status "1" ?

RavenSystem, 2019-10-14:
For "1" and "0" too. Keep in mind that your LED is an inverted relay. Just it!

mgrapiniak, 2019-10-15:
> For "1" and "0" too. Keep in mind that your LED is an inverted relay. Just it!

I have modified json to following:
{"c":{"o":1,"i":1},"a":[{"t":4,"0":{"r":[{"g":12,"v":1,"l":13}]},"1":{"r":[{"g":12,"v":0,"l":13}]},"b":[{"g":0,"p":1,"t":1},{"g":14,"p":1,"t":1},{"g":14,"t":0}]}]}

but it doesn't work. LED is OFF all the time regardless digital output of relay.
Please advise :)

John88a, 2019-10-23:
Hello, i also would like the same for the LED but i don't understand what need to change in the JSON, also what settings can be done with Eve app? i can't find any there...

RavenSystem, 2019-10-26:
Try this:

`{"c":{"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12},{"g":0,"v":1}]},"1":{"r":[{"g":12,"v":1},{"g":0}]},"b":[{"g":0},{"g":14},{"g":14,"t":0}]}]}`

closed 2019-11-04 by RavenSystem