RavenSystem/esp-homekit-devices #536: HAA Sonoff Basic + DHT22 + toggle switch

straubus, 2019-10-14:
Hi, is there any way to connect DHT22 sensor alongside external toggle switch to sonoff basic? according to github schemes it is possible but I need a template to finish this.

straubus, 2019-10-17:
Ok I connected external toggle, now i want to connect sensor. Which device type should I use to have temperature and humidity data?

bj-tracer, 2019-10-18:
The HAA hasn't TH-sensor support now ;)))
Read the Wiki for accessory types.

mientki, 2019-12-07:
Hello

I'm trying to configure Sonoff TH10 with a DTH22 sensor with HAA soft. I have a problem - the accessory tiles temperature and humidity sensor do not show measurements. JSON seems valid. What could be the reason? My configuration looks like this:

> {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":22,"g":14,"n":2},{"t":23,"g":14,"n":2},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

[image]

RavenSystem, 2019-12-07:
Use better Accessory type 24 to avoid duplicated sensor reads. And try sensor type 4 ("n":4) instead:

`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":24,"g":14,"n":4},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}`

mientki, 2019-12-08:
Excellent :) now they work perfectly. Thank you very much. Have a nice day. :)

closed 2019-12-08 by RavenSystem