RavenSystem/esp-homekit-devices #537: Ravencore - Thermostat “freezed”

claire078, 2019-10-14:
I have a device, use it for heater thermostat. 
It would work great, but once a day it freezes and values don’t updated anymore. I have to power cut (restart) and works again. 
It has an isolated relay, heater work with 230V. 
What could causes this problem?

RavenSystem, 2019-10-14:
It can be a hardware problem. Maybe a problem with the power stage.

claire078, 2019-10-15:
I just Sonoff haven’t changed yet. Sensor, cable, new programming done. 
OK, I will try it. I will come back with info.

claire078, 2019-10-16:
I have already changed. It works same. ☹️
I have no idea why.

Kristian8606, 2019-10-16:
i had this problem with my thermostat.  it may sound strange, but the problem came from hardware buttons that were realized by interruptions.  if you have an interrupt feature isolate it to test if it will freeze.

Kristian8606, 2019-10-16:
For some reason EMI is affecting the interrupt function but I still don't understand why.

claire078, 2019-10-16:
> i had this problem with my thermostat. it may sound strange, but the problem came from hardware buttons that were realized by interruptions. if you have an interrupt feature isolate it to test if it will freeze.

Interrupt function? how can I isolate it?
Remove GPIO01 button?

Kristian8606, 2019-10-16:
Comment on the interrupt and callback features in the code

claire078, 2019-10-17:
@RavenSystem : could you check it please? maybe make a test version? or with HAA it will be works 100%?

bj-tracer, 2019-10-18:
Wich GPIO you used for connection TH-sensor?
If you use any sonoff basic relay than check it for model of esp chip. New revision of relay has a esp8285 chip and GPIO2 instead GPIO14. And if you connected your TH-sensor to this pin than you have a freezes.
You need to use GPIO3 (RX-pin) and sensor will be working properly ;)))

claire078, 2019-10-18:
> Wich GPIO you used for connection TH-sensor?
> If you use any sonoff basic relay than check it for model of esp chip. New revision of relay has a esp8285 chip and GPIO2 instead GPIO14. And if you [...]

I use RX-pin and chip 8285, newer version of board.

bj-tracer, 2019-10-18:
> I use RX-pin and chip 8285, newer version of board.

Hmmmm... I had freezes when used GPIO2. But it has become properly with GPIO3.
I use sonoff on 8285 chip too.

closed 2019-10-28 by RavenSystem