RavenSystem/esp-homekit-devices #538: TH16 Thermostat

ahg675, 2019-10-15:
Hi

I have Sonoff TH16 SI7201 sensor and ravencore firmware.
Temp and humidity show 0

Can naybody help me

ps
Sorry for my english

mafyata, 2019-10-15:
There is another one issue with the same problem. Search for it.

ahg675, 2019-10-15:
Sonoff TH not supported yet ?

mafyata, 2019-10-15:
Ravencore support th10/16. I have one th16 with Sl7201 and work perfect.

mafyata, 2019-10-15:
Try to push hardly the plug to th16

ahg675, 2019-10-15:
1.
esptool.py -p COM3 erase_flash 
2.
esptool.py -p COM3 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 main.bin
3.
connect to ap and add my password
4.
adding switch to apple homekit
5,
configure switch witch eve app

dev type 5
board type 1
th sensor type 4
type 1

and temperature shows 1,0 C
humiditi 1%

On tasmota shows temperature and humiditi right value

RavenSystem, 2019-10-15:
I think that some si7021 sensors are incompatibles with esp-open-rtos driver. Better open issue there, so I can not do anything to fix it.

ahg675, 2019-10-16:
I will buy DHT11 or 
DHT22

RavenSystem, 2019-10-16:
I bought a si7021, but I send back and now I use am2301 (DHT22) for all my devices.

ahg675, 2019-10-17:
With DHT22 everything working ok :)

ahg675, 2019-10-17:
I have TH16 and DHT22 sensor. Firmware is ravencore

nmartyd, 2019-10-17:
Ah, ok, thought you were on HAA.  I will wait then.

1nutty1, 2019-10-27:
I am running a TH16 with the DHT22 with Ravencore firmware 1.2 and am getting 0 for temp and humidity...
Was working fine until opening Eve setting to check for update from 1.1, then showed 0 temp and humidity even after updating to 1.2

1nutty1, 2019-10-27:
> I am running a TH16 with the DHT22 with Ravencore firmware 1.2 and am getting 0 for temp and humidity...
> Was working fine until opening Eve setting to check for update from 1.1, then showed 0 temp and humidity even after updating to 1.2

Never mind, it would seem to be an issue in Eve. Sorted by unplugging, replugging the sensor...

closed 2019-10-28 by RavenSystem