RavenSystem/esp-homekit-devices #539: haa f0/f1 declaration

bj-tracer, 2019-10-16:
Hello!
I tried to make an interlock device yesterday.
And found a problem.
If i declarate f0 and f1 buttons for 1st accessory than it's not working propetly. I can controll Accessory from Home app but can't from fisical buttons. Other accessories at device working good.
If i declarate simple button as "b" for 1st accessory than it work good

Not working (simple example with 2 accessories) :
`{"c":{"o":0,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":14,"v":1}]},"1":{"r":[{"g":14,"v":0}]},"t":1,"s":5,**"f0":[{"g":0},{"g":10},{"g":2}],"f1":[{"g":5}]}**,{"0":{"r":[{"g":12,"v":1}]},"1":{"r":[{"g":12,"v":0}]},"t":1,"s":5,**"f0":[{"g":0},{"g":5},{"g":2}],"f1":[{"g":10}]**}]}`

Working (1st relay hasn't iterlock and use only 1 button):
`{"c":{"o":0,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":13,"v":1}]},"1":{"r":[{"g":13,"v":0}]},"t":1,"s":5,**"b":[{"g":5}]**},{"0":{"r":[{"g":12,"v":1}]},"1":{"r":[{"g":12,"v":0}]},"t":1,"s":5,**"f0":[{"g":0},{"g":10},{"g":2}],"f1":[{"g":4}]**}]}`

I have a device with 3 interlock switches and 1 self-locking switch. Want to automate it :)

RavenSystem, 2019-10-16:
You can declare only one digital input for each GPIO and press type. If you declare more, only the last will have effect.

This is a limitation of my adv-button-library. Of course, this can change in the future, because HAA try to register all, but repeated declarations only override the priors.

bj-tracer, 2019-10-16:
Ok
Noted