RavenSystem/esp-homekit-devices #543: Sonoff 4CH Pro R2 with 2 lock mechanims and 2 water valve

milonet, 2019-10-20:
Hi @RavenSystem  putting this JSON string with latest 0.3.3 firmware in my 4ch pro r2, the LED 5 become blinking to regular interval and sonoff become unreachable and cannot enter in setup mode.Only way to reuse it is flash all.

`{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":4,"i":0.4,"0":{"r":[{"g":12,"v":1}]},"1":{"r":[{"g":12,"v":0}]},"b":[{"g":0}]},{"t":4,"i":0.4,"0":{"r":[{"g":5,"v":1}]},"1":{"r":[{"g":5,"v":0}]},"b":[{"g":9}]},{"t":20,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}]},{"t":20,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}]}]}`

It's a bad json string?  I'd like to create some examples to post on your wiki.

RavenSystem, 2019-10-20:
Your JSON is right. It other bug, that I fixed just now with v0.3.4.

milonet, 2019-10-20:
Hi @RavenSystem ! I confirm that with 0.3.4 firmware i successfully setup my 4ch pro r2 with 2 lock and 2 valve.
Remain only one problem. The inching on startup. You can see this video of my sonoff: https://share.icloud.com/photos/09MqKWfoe6w2XsJHI67UH1vtQ
I add "o":1 to config in JSON, how can see the logs?

RavenSystem, 2019-10-20:
Try changing accessory orders. Declare your last accessory in the first place. 

And remove and re-pair it again int HomeKit.

milonet, 2019-10-20:
It's Work !! If you need some test ask me!

simoneorsi, 2019-10-20:
MIONET ciao,

			could you share your json string with us please.
I’ll be very helpful for me.
One more thing please. How I enter in to setup mode? I switch on the sonoff and push 10 time a r1 botton but cannot access to the system more.

Thank you so much

closed 2019-10-20 by RavenSystem

milonet, 2019-10-20:
Try press and hold for 10 seconds the button0 to enter in setup mode. The blue led blinking fast after 10 seconds.

`{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"t":20,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}]},{"t":4,"i":0.4,"0":{"r":[{"g":12,"v":1}]},"1":{"r":[{"g":12,"v":0}]},"b":[{"g":0}]},{"t":4,"i":0.4,"0":{"r":[{"g":5,"v":1}]},"1":{"r":[{"g":5,"v":0}]},"b":[{"g":9}]},{"t":20,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}]}]}`