RavenSystem/esp-homekit-devices #546: Sonoff DIY Mode complete bin file

PJDamen, 2019-10-20:
Any change of making /  getting a complete bin file instead of three seperate ones?

It would flashing with DIY mode possible I suppose.

RavenSystem, 2019-10-21:
No, sorry. My firmwares uses LCM, and there is not support for Sonoff DIY.

closed 2019-10-21 by RavenSystem

PJDamen, 2019-10-22:
I know about the LCM. But that remains intact.

It’s about the initial flash (no soldering).

I got it almost to work by reading the memory of a flashed board (initial flash with the otaboot.bin) to a bin file.

I was able to put this bin file with DIY / tasmota way into the flash memory. It started and I was able to put in my SSID and RavenSystem/haa parameters. But then at clicking join it stalls. So my bin file-size wasn’t good or corrupted.

So if you could provide the community with a proper initial bin file it would mean a whole lot of less soldering.