RavenSystem/esp-homekit-devices #548: Shelly 2.5 like Relè with external BUTTON (not switch)

milonet, 2019-10-20:
Hi, do you have a template JSON for shelly 2.5 to work like a relè with external buttons?
In ravencore the settings was:
Board Type : 5
Device Type : 2
External Toggle 1 : 1
External Toggle 2 : 1
Init State SW1 : 2
Init State SW2 : 2

It's possible make this config with HAA?

RavenSystem, 2019-10-21:
{"c":{"l":0,"i":1},"a":[{"s":5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0}]},{"s":5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0}]}]}

milonet, 2019-10-21:
It's Works! Another questions for you
1) How can enter in setup mode to edit JSON string with the string that you give me?
2) When poweron the shelly 2.5 a fast relè switching. I want try to reverse the declaration of accessory like 4ch (yesterday)

Thank you for your daily hard work

RavenSystem, 2019-10-21:
Mmm, I see GPIO 15, argh!!!!!

`{"c":{"l":0,"i":1},"a":[{"s":5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0}]},{"s":5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0}]}]}`

To enter setup mode, turn on and off any of the switches 10 times quickly (less than 1 seconds between each turn).

milonet, 2019-10-22:
It's WORK ! you can close the issue. I'm posting the correct string on homeitalia.group

Thanks!

milonet, 2019-10-22:
Hi, before my wife kills me I wanted to ask you something: D
I replaced two classic relays with shelly 2.5 and it all works with the last string you gave me. I have only one problem: very frequently I have to press the physical button more than once before the shelly turns on the light. Can it depend on the filter? how can i solve?

RavenSystem, 2019-10-22:
Ups, filter is at minimum by default. You can try to invert buttons:

{"c":{"l":0,"i":1},"a":[{"s":5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0,"i":1}]},{"s":5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0,"i":1}]}]}

milonet, 2019-10-23:
Hi @RavenSystem . Before to try your latest JSON string i want to explain the problem better:

When i push the phisical button in wall the lights not turn on immediatly, i must push the button again once.
From this moment it's need only push to turn on the lights.
If wait ~10 seconds the problem came back. I need two press to turn on the lights at first time and only single press the subsequent activations. It's a settings or a bug?

RavenSystem, 2019-10-23:
Try my last JSON. I think your buttons are inverted. Because I added "i":1.

milonet, 2019-10-23:
It's Work! You are a fckn genius !!!

RavenSystem, 2019-10-23:
Excellent!

closed 2019-10-23 by RavenSystem