RavenSystem/esp-homekit-devices #549: HAA: Motion Sensor

shinyuan, 2019-10-21:
Direct try HAA Templates from
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates
Sonoff Basic with Switch and Motion Sensor connected to GPIO 14, with status LED, and button to enter "setup mode":
{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":12,"i":10,"f1":[{"g":14}]}]}
by Using ESP8266 (NodeMCU)
.
I wire GPIO14 (D5) to VCC or GND to test the Motion Sensor,
I found that only when GPIO14 (D5) '1' --> '0' --> '1' Motion Sensor will get trigger,
but Motion Sensor (HC-SR501, HC-SR505 and RWCL-0516 ) trigger is '1' ('0' --> '1' --> '0'),
.
I add the "p" and "i" option to {"t":12,"i":4,"f1":[{"g":14,"p":0,"i":0}]} or
I add the "p" and "i" option to {"t":12,"i":4,"f1":[{"g":14,"p":0,"i":1}]} or
both GPIO14 (D5) '1' --> '0' --> '1' Motion Sensor will get trigger, 
How should I change the template?
Thanks for your help and appreciate your amazing hard work.
.
.
"f0": Sensor deactivated.
"f1": Sensor activated.
.
Buttons declaration:
Each digital input (aka button) has these options:
.
"g": Digital input GPIO:
Mandatory. GPIO connected to the digital input.
.
"p": Internal pull-up resistor:
0: Disable.
1: Default. Enable.
.
"i": Inverted digital input:
0: Default. Normal.
1: Inverted.

RavenSystem, 2019-10-21:
You should add only "i":1

RavenSystem, 2019-10-21:
Try to disable pullup resistor too with p 0.

shinyuan, 2019-10-21:
After modify template to
{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":12,"i":10,"f1":[{"g":14,"p":0,"i":1}]}]}
Why the problem still exist (the same)?
Thanks for your help.

RavenSystem, 2019-10-21:
Maybe your motion sensor is not compatible. You must use a standard motion sensor that change GPIO status when motion detected: from low to high, or from high to low.

shinyuan, 2019-10-22:
But ... why only when GPIO14 (D5) '1' --> '0' --> '1' Motion Sensor will get trigger?
Thanks for your help.

shinyuan, 2019-10-22:
Direct try only when wire GPIO14 (D5) to '1'/VCC --> '0'/GND --> '1'/VCC, Motion Sensor will get trigger?

shinyuan, 2019-10-22:
Finally, I found that
{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":12,"i":10,"f1":[{"g":14,"p":0,"i":0}]}]}
Direct try by wire GPIO14 (D5) to '0'/GND --> '1'/VCC --> '0'/GND then Motion Sensor will get trigger.

closed 2019-10-22 by shinyuan