RavenSystem/esp-homekit-devices #550: Control bistable relay with Sonoff

ondratan, 2019-10-21:
Hi guys,

I am using bistable pulse relay to control lights. There are no two-position switches, just press-buttons. It works great with RavenCore with inching time. But even if the light is on, the Sonoff shows its off of course. Do you think there is a way to inform Sonoff about current status, which would then display the correct status in the HomeKit?

Many thanks

[image]

RavenSystem, 2019-10-21:
Sorry, this is not possible with this firmware. You must use a Shelly 1 with a custom firmware to perform it, connecting the L of the lamp to Shelly S1 connector, so you can read status of your lamp.

closed 2019-11-03 by RavenSystem

vsevolod-volkov, 2019-12-02:
> Hi guys,
> 
> I am using bistable pulse relay to control lights.

Hi!

You may found the same feature request in my #577 and #618 issues.

Unfortunately Jose does want to introduce the function I propose to solve your problem to his wonderful project, so I decided to make my own branch with Action Conditions support.
You may set OTA repository as "**vsevolod-volkov**/haa" instead of "**RavenSystem**/haa" regarding to [Jose's installation instructions](https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation) to install firmware with action conditions support in format as shown in #618.
Please find configuration example debugged on my own Shelly 1 used as pulse relay controller below:
```
{
    "c": {
        "o": 1,
        "b": [ { "g": 0, "t": 5 } ]
    },
    "a": [
        {
            "t": 1,
            "0": { "r": [ { "c": [{ "g": 5, "t": 0 }], "g": 4, "v": 1, "i": 0.1 } ] },
            "1": { "r": [ { "c": [{ "g": 5, "p": 0 }], "g": 4, "v": 1, "i": 0.1 } ] },
            "f0": [ { "g": 5, "p": 0 } ],
            "f1": [ { "g": 5, "t": 0 } ]
        }
    ]
}
```
As you can see "a"..."r"..."c" element format uses "g" "p" and "t" subelements as described for button. "t" is selected as input GPIO state to whole condition element to look like a button setup. Note: if you use some GPIO in condition's "g", that has no button attached to it, that condition should be the first one.

ondratan, 2019-12-08:
Hi vsevolod-volkov, 
I appreciate your passion for the thing, you did a great job. I will definitely try it. I also read Joses comments and I hope he takes it at his mercy to implement this feature as the use of bistable relays increases :)

vsevolod-volkov, 2019-12-09:
Hi ondratan,
I do it for myself, it is my own problem too (please find #577).

vsevolod-volkov, 2019-12-09:
> Please find configuration example debugged on my own Shelly 1 used as pulse relay controller below:

Updated config:
```
{
    "c": {
        "o": 1,
        "b": [ { "g": 0, "t": 5 } ]
    },
    "a": [
        {
            "t": 1,
            "1": { "r": [ { "c": [{ "g": 5, "t": 0 }], "g": 4, "v": 1, "i": 0.1 } ] },
            "0": { "r": [ { "c": [{ "g": 5, "p": 0 }], "g": 4, "v": 1, "i": 0.1 } ] },
            "f0": [ { "g": 5, "t": 0 } ],
            "f1": [ { "g": 5, "p": 0 } ]
        }
    ]
}
```

vsevolod-volkov, 2019-12-09:
Also please refer to #648.

vsevolod-volkov, 2019-12-15:
> I will definitely try it.

Please find [new OTA binary release ](https://github.com/vsevolod-volkov/haa/releases/tag/v0.8.3.ac) including input GPIO filtering plus all the features from current latest original HAA release v0.8.3.

To use input GPIO filtering add "F" option to "c":
```
{
    "c": {
        "o": 1,
        "b": [ { "g": 0, "t": 5 } ],
        "F": 10
    },
...
```