RavenSystem/esp-homekit-devices #554: External sensor for Thermostat

ptath, 2019-10-23:
I probably know all yours arguments, but is it a way to assemble HAA thermostat with wireless (remote or any other source) temperature/humidity sensor?

My first thermostat on ravencore perfectly worked 6 months, but now in living house I cannot use only one wired sensor. Right now it is nodered and 5 Zigbee sensors, so thermostat gets rounded temps of these 5.

PS Thanks for a great product, using it more than 1 year =)

RavenSystem, 2019-10-23:
@ptath As you know, actually that is impossible. But I want to add http-get method. HAA would work like a http client, and you could use an external temperature sensor that serves data using a web server.

ptath, 2019-10-23:
> But I want to add http-get method

It will be great!

Another thing I wish in my dreams it is serialsend for devices like LC Technology relays (2-8 channels relay on esp8266 chip, esp01s but controls relays with serial commands like `A00101A2`), I use one for irrigation:

[image]

RavenSystem, 2019-10-23:
Actually, there is only one type of output, gpio digital outputs, but I want to add others, like IR commands, http-get, and serial could be other.

Partizan7676, 2019-10-24:
> I probably know all yours arguments, but is it a way to assemble HAA thermostat with wireless (remote or any other source) temperature/humidity sensor?
> 
> My first thermostat on ravencore perfectly worked 6 months, but now in living house I cannot use only one wired sensor. Right now it is nodered and 5 Zigbee sensors, so thermostat gets rounded temps of these 5.
> 
> PS Thanks for a great product, using it more than 1 year =)
> 
> Immediately after pressing "submit" button I've got an idea of some kind of bridge between HAA devices, bypassing HomeKit (like mqtt, yes). Maybe some kind of webhooks or any input other than just physical pins on device.

you can use automation with homekit. create it with eve app. i do that for electric heater(two in one room). one with thermostat, other with s20 outlet. when thermostat start heating socket give power two second heater.

ptath, 2019-10-24:
> you can use automation with homekit. create it with eve app. i do that for electric heater(two in one room). one with thermostat, other with s20 outlet. when thermostat start heating socket give power to second heater.

Thank you, but how to send remote sensor data to the first thermostat? =)

offtop: Меня тоже достал газовый котёл, не дает больше 40 градусов в кране и похоже единственный вариант — запараллелить его с обычным электрическим водонагревателем. В этом случае отлично подойдет эта схема, уже прошил s20 под это дело

RavenSystem, 2019-10-24:
@Partizan7676 Great idea. 

@ptath You don’t need to send sensor data. Your remote sensor is your thermostat.

mafyata, 2019-10-25:
I am using nodemcu with dht22 sensor.

Partizan7676, 2019-10-31:
> > @Partizan7676 Great idea.  ...
> 
> @RavenSystem thanks, nice workaround but it need 2 devices, one of them dedicated and ac-powered esp8266 with sensor.
> 
> Anyway the possibility of using external data will be great )
> 
> Sent with [GitHawk](http://githawk.com)

external sensor? remember plugin for Broadlink remote - they found solution only with MQTT. other ways increase ROM size and load of ESP...

closed 2019-11-29 by RavenSystem

ptath, 2020-02-01:
Thanks for the HomeKit Rest API and ping inputs, great job. Works perfects with simple on/off needs.

Back to my question, `curl -X PUT -d '{"characteristics":[{"aid":1,"iid":10,"value":20}]}' http://192.168.1.233:5556` has no effect because we cant change thermostat temp in this way, right? =)

Can we somehow avoid this restriction?