RavenSystem/esp-homekit-devices #556: HAA - Shelly 2.5 - how reset ??

Fudalii, 2019-10-24:
Hello.

How to reset the device to re-enter WiFi or new JSON data?

The Shelly 2.5 template was entered.. Everything works ok. But I can't reset it.

Within 1 minute I push a small button 10x and I see 3 x led flash. But the HAA network doesn't appear

RavenSystem, 2019-10-24:
You must hold build-in button for 10 seconds, or trigger 10 times any of switches.

Fudalii, 2019-10-24:
Welcome.

I tried it (like in RavenCore) but it doesn't help.

I tried the external button and the small button on the housing.

The reset is confirmed correctly (3 flashes of the LED) but the device does not reset. I can still add them to HomeKit.

I also tried to upload Flash again but it's the same :)

RavenSystem, 2019-10-25:
HAA can not be reset. You can enter in setup mode to reconfigure it.

bj-tracer, 2019-10-25:
> The reset is confirmed correctly (3 flashes of the LED) but the device does not reset.

You need connect to HAA-xxxxxx access point and configure wi-fi connection/json string/other options.
And you can reset wi-fi/homekit settings at this mode.

Fudalii, 2019-10-25:
Well. I understand. So in the template on the Wiki page I have to add this option for GPIO button on the housing ... ?

RavenSystem, 2019-10-26:
I have updated Shelly2.5 template to add "setup mode" button. Hold it for 10 seconds and device will restart in "setup mode".

https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#shelly25-switch-with-external-toggles-and-status-led

Fudalii, 2019-10-26:
WoW ... Thx !

closed 2019-10-26 by Fudalii

dan003400, 2023-02-05:
> You must hold build-in button for 10 seconds, or trigger 10 times any of switches.

This doesn't work. Tried 10 times, toggling either SW1 or SW2 does absolutely nothing.

StefanoGiuli, 2023-02-07:
new version HAA  Home Manager v3.4.4 now work (i set my ipad on 2.4 ghz wifi)