RavenSystem/esp-homekit-devices #559: Nodemcu + Temp/hum

mafyata, 2019-10-25:
Hi , I have nodemcu and dht22 for temp and humidity.
If I use haa and want only to show emp and humidity(type 24), what template to use ?
Is it correct:
{
  "c": {
    "o": 0
  },
  "a": [
    {
      "t": 24,
      "g" : 14,
      "z": -1
    }
  ]
}

RavenSystem, 2019-10-25:
Yes.

mafyata, 2019-10-25:
Can I enter setup mode with this template?

RavenSystem, 2019-10-25:
Ops, no. You must declare a way to do that. With a physical button or with a dummy switch.

mafyata, 2019-10-25:
I don’t want switch in home app. I think to enter setup mode with connecting pins with jumper.

RavenSystem, 2019-10-25:
That is the best way. Dummy switch sucks, IMO, but I tell you all options.

mafyata, 2019-10-25:
IMO?

RavenSystem, 2019-10-25:
In My Opinion.

mafyata, 2019-10-25:
I think for using GPIO 1 and GND for setup mode. Can you tell me the json with them for setup mode.

RavenSystem, 2019-10-25:
Sure, what press type do you want? Single, double, holding for 10 seconds...

mafyata, 2019-10-25:
single

RavenSystem, 2019-10-25:
`{"c":{"b":[{"g":1,"t":0}]},"a":[{"t": 24,"g":14,"z":-1}]}`

Be sure that gpio1 works as expected in node mcu.

mafyata, 2019-10-25:
Here is a picture with all pins on nodemcu
[image]

RavenSystem, 2019-10-25:
Gpio1 is uart tx. Better use other. 5 is good.

mafyata, 2019-10-25:
Gpio 6?

RavenSystem, 2019-10-25:
Don’t use 6, 7 and 8.

mafyata, 2019-10-25:
ok , I will use 5.
{"c":{"b":[{"g":5,"t":0}]},"a":[{"t":24,"g":14,"z":-1}]}
Is this correct?

RavenSystem, 2019-10-25:
Yes.

mafyata, 2019-10-25:
Works perfect. Thanks. How to update the firmware in future? With entering in setup mode only ?

RavenSystem, 2019-10-26:
“ If you installed it using OTA procedure, there is an option called "Auto OTA Updates" that allows to search updates every time device enters in setup mode, 90 seconds pasts, and setup web page is not loaded.”

https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect#setup-mode

mafyata, 2019-10-26:
Yes , I selected auto ota update. So to update I must enter the device in setup mode , it will auto update and restart itself to normal mode ?

RavenSystem, 2019-10-26:
Exactly, always you don’t load setup mode web page.

closed 2019-10-26 by RavenSystem