RavenSystem/esp-homekit-devices #561: Questions: Sonoff SV + HAA

claire078, 2019-10-26:
Sonoff SV already supported with HAA firmware?

RavenSystem, 2019-10-26:
All devices with an esp8266/esp8285 chip are supported.

claire078, 2019-10-26:
The templates are labeled in HAA templates. Sonoff mini, Jelly, etc. 
That is why I asked it. Then just because the IO names are different?

RavenSystem, 2019-10-26:
What do you ask with "Then just because the IO names are different?"? I don't understand question.

claire078, 2019-10-26:
Why do templates have labels Sonoff mini, Jelly etc. ?

RavenSystem, 2019-10-26:
Because they use GPIOs of labeled devices.

claire078, 2019-10-27:
Possible to enter setup mode with Homekit button without trigger the relay? For example push and hold the button in Home app without switch the relay. 
I have some devices wich are not easy to access.

RavenSystem, 2019-10-27:
No, HomeKit is not prepare to send push and hold events from Apps. If you want to enter setup mode without accessing device and without trigger relay, you must declare an additional HomeKit switch without actions, so you can use it to enter setup mode toggling quickly it. Because HAA supports multiroom, you can move that "dummy" switch to other room.

claire078, 2019-10-27:
Every device with one dummy switch? 😳
In action mode (not setup mode) could able to switch device with its ip address (webpage) to setup mode?

RavenSystem, 2019-10-28:
No, sorry. You must see device IP address in your DHCP server.

closed 2019-11-07 by RavenSystem