RavenSystem/esp-homekit-devices #562: Setup process not clear

Dirk23, 2019-10-26:
Hello,

I bought several Sonoff Dual R2. I flashed the Sonoff Dual with new firmware according to this manual: https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation. After I added the Sonoff to my WLAN and entered the OTA repository, I can't use the Sonoff Dual.
into HomeKit. Also 10x the integrated key or the button 0 press brought no change. I don't know what to do to insert the Sonoff Dual into HomeKit according to the instructions. If they are already available in HomeKit after the integration into the WLAN and the OTA setup, or you can only record the devices in HomeKit after you have worked through this point: https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation#homekit-setup

Regards

Dirk

RavenSystem, 2019-10-26:
What repository have you use? HAA?

Dirk23, 2019-10-27:
I setup „RavenSystem/haa“ as OTA repository in Captive Portal

Dirk23, 2019-10-27:
I just found out by chance, that after the first setup via the Captive Portal a new WLAN with a new Captive Portal was created on the Sonoff, in which I should now enter a JSON Config. This step was completely missing in the setup process. I am now looking for a JSON template for a Sonoff Dual R2, which should be used to control the blinds.

Dirk23, 2019-10-27:
What I don't know is if the new WLAN, whose name starts with HAA, just came because I pressed „button 0“ 10 times or if it just appeared that way.

RavenSystem, 2019-10-27:
Please, read the Wiki: https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation#homekit-setup

"Once you configure your JSON string in Home Accessory Architect firmware using its Setup Mode, you will can pair your device with HomeKit."

Dirk23, 2019-10-27:
Yes, i read that. Nur there is no Info, that you Need to Access the new Wifi Named „HAA...“ to configure that JSON String. Is there a template for Sonoff Dual R2 somewhere, because i dont understamd that JSON Stuff

RavenSystem, 2019-10-27:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect#setup-mode

"... Once device is in setup mode, you can access to a webGUI connecting to device IP address (in case a wifi is configured) or connecting directly to a wifi AP generated by device."

https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect

And as you can see in the Wiki, there is not any HomeKit Accessory Type to control blinds yet.

Dirk23, 2019-10-27:
So do i Need use Ravencore until haa is able to Control Blinds?

RavenSystem, 2019-10-27:
Yes.

Dirk23, 2019-10-27:
Is there a was from ravencore to haa without flashing all Sonoffs again?

pilot1981, 2019-10-27:
Yes, you have to update to last version of RC: you will find new feature to upgrade to HAA

closed 2019-11-07 by RavenSystem