RavenSystem/esp-homekit-devices #563: Leak detector operation like a toggle switch

pawelosin, 2019-10-27:
Hi, I would like to use HAA firmware with Wemos D1 mini and Hunter Mini-Clik rain sensor as a leak sensor. But the rain sensor works like a toggle - the circuit is opened or closed. Is there  possibble to configure the HAA  to work when the circuit is opened the leak is detected and when the circuit is closed the leak is not detected?

RavenSystem, 2019-10-27:
Yes, you can.

closed 2019-10-28 by RavenSystem

pawelosin, 2019-11-03:
There is my template to Wemos D1 mini with switch, temperature sensor and leak sensor:
{"c":{"l":2,"i":1,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":22,"n":3,"g":5},{"t":7,"f1":[{"g":14,"t":1}],"f0":[{"g":14,"t":0}]}]}