RavenSystem/esp-homekit-devices #565: HAA Templates for H801

kutdenis, 2019-10-27:
Now running on ravencore, wanted to upgrade to HAA. Is there a working configuration on this device?

https://github.com/arendst/Tasmota/wiki/H801

RavenSystem, 2019-10-28:
HAA does not support RGBW lights yet.

RavenSystem, 2019-11-07:
Last version of HAA supports RGBW lights. Can you provide info about your RavenCore setup, please?

kutdenis, 2019-11-08:
Board type 1, Device type 14, Pin B 12, Pin G 13, Pin R 15, Pin W 14

RavenSystem, 2019-11-08:
`{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"w":14,"b":[{"g":0}]}]}`

RavenSystem, 2019-11-08:
Ensure that you are using last HAA version (v0.6.1).

kutdenis, 2019-11-11:
Can't go into setup mode now. Removed and updated to 0.6.6 through the computer, configured the new and installed back. But can't update over the air to 0.6.7. Also not included in the setup mode. Entered the setup mode as specified in the wiki

kutdenis, 2019-11-11:
Sonoff switches are perfectly updated

RavenSystem, 2019-11-11:
How are you trying to enter in setup mode?

kutdenis, 2019-11-11:
10 taps in the home app. It works with sonoff

RavenSystem, 2019-11-11:
0.6.6 is less responsive with lightbulbs. Try doing it slower. A little less than a second between each tap.

kutdenis, 2019-11-11:
Updated, very slowly clicked 10 times!

RavenSystem, 2019-11-11:
This works better with last version.

closed 2019-11-11 by RavenSystem

kutdenis, 2021-05-10:
At some point, the device stopped responding, tried to flash it, but it still does not respond. Maybe the template has changed?