RavenSystem/esp-homekit-devices #567: HAA

netRunner0, 2019-10-28:
Hello!

Please add a type of intercom with the ability to select gpio to call (short, long press)

Thanks

RavenSystem, 2019-10-28:
Intercom? Do you mean video doorbell?

netRunner0, 2019-10-28:
Yes:

netRunner0, 2019-10-28:
[image]

RavenSystem, 2019-10-28:
That is totally impossible. ESP8266 chip can not manage a video camera.

netRunner0, 2019-10-28:
No camera needed. The main thing is the notification intercom

RavenSystem, 2019-10-28:
You can not make a HomeKit doorbell without a camera. It is mandatory.

netRunner0, 2019-10-28:
can I take a static screenshot?

RavenSystem, 2019-10-28:
No, you can not. ESP8266 has not enough power to manage and process images.

RavenSystem, 2019-10-28:
A good alternative could be to use HomeKit occupancy sensor accessory type.

netRunner0, 2019-10-28:
Is it possible to implement on esp32?

RavenSystem, 2019-10-28:
Yes.

netRunner0, 2019-10-28:
I would like to see this feature in future releases of the HAA)

Thanks for the quick answers ;)

closed 2019-11-01 by RavenSystem