RavenSystem/esp-homekit-devices #568: Thermostat json on TH16

Codi878, 2019-10-28:
Hi Jose,

I have a problem with the correct thermostat configuration - json file. 

I would like it to be visible in Homekit, just like in Ravencore, and now I have a switch + a thermometer and humidity separately. 

How this config should look like now ?

Thank You

My json:

{
  "c": {
    "l": 13,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1
          }
        ]
      },
      "b": [
        {
          "g": 0
        }
      ]
    },
    {
      "t": 21,
      "g": 14,
      "m": 12,
      "x": 28,
      "d": -0.2
    }
  ]
}

hexardavies, 2019-11-01:
I'm interest to know how you went with this fix... Json works / and have tried many other ways - can not get it to work the relay!!!

caligari74, 2019-11-01:
You can find a template here:
https://github.com/RavenSystem/esp-homekit-devices/issues/560

Codi878, 2019-11-01:
Hi 
I check this template and is not good for me becouse switch is not assosiated with relay of thermostat.

I would like the thermostat to be connected to the relay as one.

hexardavies, 2019-11-01:
Yeah I have been working today on trying to get the relay on the board to be controlled by the temp. (Eg. AV cabinet temp fan controller)

RavenSystem, 2019-11-01:
I wrote a sample:

https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#sonoff-th1016-heater-thermostat-with-dht22-sensor

Codi878, 2019-11-01:
Thank you very much Jose you are amazing. I will check this example ☺️

RavenSystem, 2019-11-01:
It works like a charm. That JSON is used with a TH16 and a heater in my house.

hexardavies, 2019-11-01:
Quick question how do you with this code - to go back to setup mode... or resetting the wifi.

RavenSystem, 2019-11-03:
To enter setup mode you must press and hold built-in button of Sonoff TH more than 10 seconds, until you see LED blinks. Or turn on and off thermostat quickly more than 10 times from HomeKit.

closed 2019-11-07 by RavenSystem